Dioptas/Dioptas #6: ImportError: No module named CifFile

picca, 2016-03-23:
Hello,

when I try to run diopats from the source directory on my Debian stable system, I get this error
Did you forget to commit this CifFile module ?

test@diffabs6:~/dioptas/Dioptas/dioptas$ python Dioptas.py 
No module named smooth_bruckner
WARNING:model.util.BackgroundExtraction:Could not import the Fortran version of smooth_bruckner. Using python implementation instead. Please run 'f2py -c -m smooth_bruckner smooth_bruckner.f95' in the model/util folder for faster implementation
Traceback (most recent call last):
  File "Dioptas.py", line 25, in <module>
    from controller.MainController import MainController, get_version
  File "/home/test/dioptas/Dioptas/dioptas/controller/__init__.py", line 19, in <module>
    from .CalibrationController import CalibrationController
  File "/home/test/dioptas/Dioptas/dioptas/controller/CalibrationController.py", line 25, in <module>
    from widgets.CalibrationWidget import CalibrationWidget
  File "/home/test/dioptas/Dioptas/dioptas/widgets/CalibrationWidget.py", line 24, in <module>
    from widgets.plot_widgets import MaskImgWidget, CalibrationCakeWidget
  File "/home/test/dioptas/Dioptas/dioptas/widgets/plot_widgets/__init__.py", line 5, in <module>
    from SpectrumWidget import SpectrumWidget
  File "/home/test/dioptas/Dioptas/dioptas/widgets/plot_widgets/SpectrumWidget.py", line 24, in <module>
    from model.util.HelperModule import calculate_color
  File "/home/test/dioptas/Dioptas/dioptas/model/__init__.py", line 24, in <module>
    from PhaseModel import PhaseModel
  File "/home/test/dioptas/Dioptas/dioptas/model/PhaseModel.py", line 22, in <module>
    from model.util.cif import CifConverter
  File "/home/test/dioptas/Dioptas/dioptas/model/util/cif.py", line 25, in <module>
    from CifFile import ReadCif
ImportError: No module named CifFile

Luindil, 2016-03-23:
Hei,

there is a new dependency called "PyCifRW" which provides the CifFile library. Please just use

pip install pycifrw

and it should work

picca, 2016-03-25:
Ok

closed 2016-03-25 by picca