Dioptas/Dioptas #10: Impossible to run Dioptas

jambondeparis, 2016-05-11:
Hello Clemens

When I try to run Dioptas, I have an error message. (I am on W7 64bits)
This is the log : 
--------------------------------------------------------------------------------
2016-05-11, 10:25:09
--------------------------------------------------------------------------------
<type 'exceptions.ValueError'>: 
dictionary update sequence element #12 has length 1; 2 is required
--------------------------------------------------------------------------------
  File "<string>", line 94, in <module>
  File "dioptas\controller\MainController.py", line 86, in __init__
  File "dioptas\controller\MainController.py", line 211, in load_settings
  File "dioptas\controller\MainController.py", line 221, in load_directories
--------------------------------------------------------------------------------
Dioptas Version: 0.3.1

It's weired, I have used Dioptas 1 year and now no version works (0.2.3, 0.3.0*)

Thanks for your help

Jean-Antoine

Luindil, 2016-05-11:
Hei Jean-Antoine,

could you please delete the ".Dioptas" folder in you user directory (something like "c:/users/jean_antoine/" and try again?

Best wishes,
Clemens

jambondeparis, 2016-05-11:
**Yes !!!**

Thank you Clemens for this quick response.

Best regards
Jean-Antoine

closed 2016-05-11 by jambondeparis