Dioptas/Dioptas #15: Inconsistancy between image and mask

vallsv, 2016-06-13:
Hi, I try to use Dioptas from master and pyFAI 0.12.

- On calibration tab
- Load a non squre image: i loaded ```Pilatus1M.edf``` from pyFAI unittests
- Rotate +90

```
File "/home/valls/workspace/dioptas.git/dioptas/controller/MainController.py", line 159, in tab_changed
    self.integration_controller.spectrum_controller.image_changed()
  File "/home/valls/workspace/dioptas.git/dioptas/controller/integration/PatternController.py", line 152, in image_changed
    x, y = self.calibration_model.integrate_1d(mask=mask, unit=self.integration_unit, num_points=num_points)
  File "/home/valls/workspace/dioptas.git/dioptas/model/CalibrationModel.py", line 325, in integrate_1d
    filename=filename)
  File "/usr/lib/python2.7/dist-packages/pyFAI/azimuthalIntegrator.py", line 2399, in integrate1d
    unit=unit, split=split)
  File "/usr/lib/python2.7/dist-packages/pyFAI/azimuthalIntegrator.py", line 1157, in setup_CSR
    assert mask.shape == shape
```

Maybe pyFAI 0.12 provides more sanity checks from input params.

vallsv, 2016-06-13:
I forgot to say that the content of shape and mask.shape was:
```
shape = (1043, 981)
mask.shape = (981, 1043)
```

Luindil, 2016-06-13:
Hi,

I honestly did not check yet with pyFAI 0.12  does it work with the "before_refactoring" branch?

Did you create the mask before or after rotating the image?

vallsv, 2016-06-13:
With the branch "before_refactoring" it's exactly the same:

```
  File "/home/valls/workspace/dioptas.git/dioptas/controller/integration/PatternController.py", line 152, in image_changed
    x, y = self.calibration_model.integrate_1d(mask=mask, unit=self.integration_unit, num_points=num_points)
  File "/home/valls/workspace/dioptas.git/dioptas/model/CalibrationModel.py", line 325, in integrate_1d
    filename=filename)
  File "/usr/local/lib/python2.7/dist-packages/pyFAI/azimuthalIntegrator.py", line 2428, in integrate1d
    unit=unit, split=split)
  File "/usr/local/lib/python2.7/dist-packages/pyFAI/azimuthalIntegrator.py", line 1157, in setup_CSR
    assert mask.shape == shape
```

I never created any mask. I guess an empty mask is created by default?

Luindil, 2016-06-13:
By Default the mask should be None.

I tried to reproduce this error, but am not able to do it with neither python 2 nor python 3. 

Even when creating a mask and and then rotating the image it does just erase the complete mask and no error pops up.  Can you please delete ~/.Dioptas and try again?

However, I realized that there are some *.poni files in the pyfai/testimages folder which do not have wavelengths, which will cause an error when loading with Dioptas, because it assumes that there is always a wavelength present...

Another error with the Dioptas master branch and pyFAI 0.12 is when you click in the image in the integration view, which is however fixed in the develop branch of Dioptas.

Luindil, 2016-06-13:
actually it is not right that the mask is None, but everytime an image is loaded it will be checked that the dimensions are the same.

this is done in the MaskController line 64.

vallsv, 2016-06-14:
Hi. I reach deep into that.

I moved the ```Pilatus1M.edf``` into another empty location. But Dioptas was still loading ```Pilatus1M.poni``` at launch time from the previous location. I dont check why.

As result ```self.spectrum_geometry.mask``` was initialized.

And when it calls ```integrate1d(mask=None)```, the mask used is the one from the azimuthal integrator attribute.

Maybe Dioptas can initialize its own mask with ```self.spectrum_geometry.mask``` if it exists, and then force the attribute to ```None```. But the rotation to 90 is still, i think, not easy to manage cause there is also pyfai detector modelization.

Anyway as you said, from a fresh load, i have no problems with rotations.

Luindil, 2016-06-20:
So does it work now? Can I close the issue?

vallsv, 2016-06-20:
Depend of what you want to aim.
- If you load a poni file which contains a mask it will not working
- If you use a detector which contains a mask it will not working (i guess)
- If you only want to support your own mask, it is working without any problem
Then it is up to you.

Luindil, 2016-06-20:
Ok, right now, Dioptas is aimed as standalone and does not work in combination with pyFAI scripts. This is already due to the choice of geometry. My initial choice was to follow Fit2D convention which pyFAI did not do. The main reason for this was, that i needed to convince people who were using Fit2d for over 20 years... Therefore calibration poni files generated with pyFAI will have wrong geometry...
That means people using Dioptas right now just use poni files generated with Dioptas, which do not have detector or mask specified in the poni file.

The major question right now, is what you are going to do with the GUI for pyFAI. When I was last time ESRF, in october 2015, Jerome was proposing to include Dioptas as GUI into pyFAI. I am certainly up for this. And if this is still wanted then we would have to think about these issues

What is the current status of the discussion? If I understood it correctly, you are the one who is hired for the GUI part.

Best wishes
Clemens

vallsv, 2016-06-20:
Sorry, i would have present myself before. I was thinking it was a very common issue.
Then yes, i was hired to work on the GUI for pyFAI, and right now i am only doing a state of the art of azimuthal integration GUIs.

Jerome explains to me that merging Dioptas was one of the solutions. And i think your software is a good candidate cause the source code looks well designed and the application looks clean. But we never seriously talk about that, and what will be the consequences on the features and architecture of your application. I think we have to clarify what we want to do first. I will ask Jerome.

closed 2017-05-20 by CPrescher