Dioptas/Dioptas #17: Dioptas Windows 10 Compatability

shunter8, 2016-07-24:
Hello,
Sorry if this is the wrong place to post this.
However, I wish to ask if Dioptas is compatable with Windows 10?
I recently downloaded Dioptas v0.3.1 and installed Anaconda and configured it as described on the main page.
However - after running Dioptas.exe it opens normally, but as soon as I try to load a mar image filed the programs stopps working and crashes.
In addition to v0.3.1 I have also tried older versions, with the same result.

Is this an incompatability with Windows 10?

Best REgards,
Steve.

Luindil, 2016-07-25:
Dear Steve,

this is certainly the right place to post this, since there is no other forum currently available.

I currently develop Dioptas under Windows 10 and do not have problems when running it from source. One thing I want to note: In principle you do not need Anaconda if using the executable version of Dioptas, but I probably have to specify this more explicitly in the instructions.

I could somehow reproduce your error with the 0.3.1 executable version and found a workaround for now. Can you please delete the "C:\Users\Username\.Dioptas" folder before starting Dioptas and try again? This was working for me, however, i had to do it before every start of Dioptas. I am not sure yet what is causing this problem.

Another possibility is to run it from source. If you anyway installed all libraries in anaconda already, you can just download the source code from the github repository and run it either from the anaconda command prompt by cd into the Dioptas\dioptas directory and run "python Dioptas.py". Or create a small .bat script with something like:

"C:\Anaconda3\pythonw.exe" "C:\Users\Clemens Prescher\Programming\Python\Dioptas\dioptas\Dioptas.py"

Where the paths should be adjusted to your anaconda installation and the directory where the source files are.

Another major issue could be the memory of your machine. Mar345 files are very big, and since Dioptas is using a lookup table implementation for the integration (therefore the speed is relatively fast). I would recommend at least 8 GB of RAM to be able to run it. 4 GB might be already the limit where it does not work.

I am sorry for the inconvenience and will try to work on the problem.

Best wishes,
Clemens

shunter8, 2016-07-25:
Thanks very much for your reply!

A bit of background about my system
I have an Intel Core i5-3320M CPU @2.6 GHz, * GB of RAM and a 64-bit OS (Windows 10 Pro).

It turns out that somehow my calibration image file is corrupt. I sent the image to my colleague (who has no problems opening calibration files) and when attempting to open the image file, their program also stopped working and crashed.

Nevertheless, I have still found issues with the executable v0.3.1 in addition to the source code version.

I am performing the following:
Either open executable, or run python script.
In Calibration tab, Load Calibration then select .poni file.
Go to Integration tab, then Load Image(s).

Using the Dioptas v0.2.4 executable - I can perform all of the above steps without issue - the image opens no problem, is integrated, additionally the autoprocess and autocreate functions in the Image and Patterns tabs work perfectly.

Using the Diotas v0.3.1 executable, i can load the calibration poni file with no problems. Switching to the Integration tab is fine, however when I Load Image(s) and choose a file the program becomes unresponsive (Not Responding). If I leave it in this state, after about 2-3 minutes, an OOOPS! An error occurred! popup appears (A screenshot is attached - filemane Dioptas031-error.png). After clicking OK the image then loads, however no integration of the image is performed.
This happens regardless of whether or not I delete "C:\Users\Username.Dioptas" folder before starting Dioptas.

I get even more problems when trying to run the Dioptas Python script (I suspect that this is due to my systems python being incorrectly set-up, rather than a problem with dioptas.py!).
When launching the dioptas python script, I get 3 warnings as shown in the screenshot python-cmd-wind-start-script.png
The calibration .poni file loads ok, a screenshot of the command windows is shown in python-cmd-wind-af-load-calib.png
However, as soon as i try to swith to the Integration tab (just clicking the tab, I do not even get to the stage of trying to load and image) an OOOPS! An error occurred! popup repetively appears (after clicking OK, it pops back up.
A screenshot of the error message is attached as python-error-switch-int-tab.png and the corresponding command window can be found in python-cmd-wind-swith-int-tab.png
The only way to exit is to kill the command prompt window.

As I said, I suspect this issue is on my end rather than dioptas - I just included as much detail as I could for completeness.

The Dioptas v2.0.4 executable works fine, and so I will continue to use that for now.

Thanks very much,
Steve.

[image]
[image]
[image]
[image]
[image]

CPrescher, 2017-05-20:
This is actually an error in pyqtgraph, if you upgrade pyqtgraph it should work again without problem.

Best,
Clemens

closed 2017-05-20 by CPrescher