Dioptas/Dioptas #24: save combined pattern when using configurations

Luindil, 2016-11-29:
the save pattern button, currently saves the selected configuration but not the combined pattern, when selected. This is confusing for the user.

Should there be an autosave for the combined pattern?

mlsims, 2016-12-30:
I don't think it's the same question, but I was wondering if there's a way to batch save all the images in the folder as patterns at the same time. Saving them one by one is slow when I have a few thousand files to do. It's a really helpful program, so thank you for your response!

Luindil, 2016-12-30:
hi,
of course there is a batch mode. Just select all image files you want to integrate in the file browser after you have clicked the "Load Image(s)" button. When you already chose a folder for automatic saving of patterns it will save all the integrated patterns in this folder, if you have not yet chosen a folder, you will have to select one. Dioptas will then batch integrate all images and save the patterns. All corrections you have currently selected (like mask) will be applied to all patterns. Also background subtracted patterns will be saved, if it is selected.

schoelme, 2019-01-16:
Hi Clemens, 
is there now a possibility to use autosave for a combined pattern (e.g. for the MEC detector setup)? I didn't see it for 4.0.0? It is a great programme you have developed! Thanks for your work!

CPrescher, 2019-01-16:
Hi, 

unfortunately not yet, but I am working on it...

CPrescher, 2021-09-26:
added this feature in commit 587b6854d03e06d1a248d8ccafcb0d9560d5ac9c

closed 2021-09-26 by CPrescher