Dioptas/Dioptas #36: pyFAI issue

ClundXIII, 2017-04-21:
Hi,

I installed the custom pyFAI dependency for Dioptas:
$ locate pyFAI
[...]
/home/simon/pyFAI/Cython-0.25.2-py2.7-linux-x86_64.egg/

It is not installed anywhere else. (I removed it via pip and anaconda)

Dioptas therefore failed to start:
<type 'exceptions.ImportError'>: 
No module named pyFAI.massif
(etc.)

I was able to workaround this problem by exporting the python lib path:
export PYTHONPATH="/usr/local/lib/python2.7/dist-packages/pyFAI-0.14.0a3-py2.7-linux-x86_64.egg/:$PYTHONPATH"

this worked and I was able to start Dioptas.

CPrescher, 2017-05-02:
Hi,

I think this is a system specific issue and has something to do with your specific Pythonpath configuration. Therefore, I am not able to comment on it or give a comprehensive solution. Usually this happens due to several python installations. In principle I recommend do just use anaconda python an unix based system, since this does not have an effect on the system python.

Best wishes,
Clemens

ClundXIII, 2017-05-02:
Hi,

I am using Anaconda.
Also, as you noted in your README, Anaconda does not contain the correct pyFAI version.

> The only more advanced dependency is to install pyFAI, since the current release version has some problems with the newest scipy version, please go to a temporary folder pyand run:

Dioptas was just not able to find the pyFAI installation in my case. I will try to find the reason and make a bug report for Ubuntu / Linuxmint if needed.
You might want to add my workaround to your installation instructions in case I am not the only one affected by this problem.

Regards,
Simon

CJ-Wright, 2017-05-13:
pyFAI is on [conda-forge](https://github.com/conda-forge/pyfai-feedstock), you might try to install it from there. If there are conflicts between the packages please let us (the pyFAI feedstock maintainers) know.

ClundXIII, 2017-05-13:
@CPrescher might be able to answer this to you since he inserted the specific part into the installation instructions about using a fork of pyFAI:
https://github.com/Dioptas/Dioptas/commit/231d81633f7012a56f5cf15f0787d75db49b3ed6

I kindof believe that this is just my local configuration right now. It wouldn't be the only PATH related problem that is broken on my linux. Feel free to close this issue if you want.

CPrescher, 2017-05-13:
I also suspect it has something to do with your linux paths, because I regularly reinstalled Dioptas from scratch and never had this kind of problems.

I actually updated the Readme, just right now, with installation instructions for only using conda-packages, maybe that will suit your needs.

Best wishes,
Clemens

closed 2017-05-13 by CPrescher