Dioptas/Dioptas #37: Doesn't run in MacOS

jglezplatas, 2017-04-27:
When I try to run Dioptas program I have the next message error:

Error of OSStatus -67062

erangre, 2017-04-30:
I am able to run the mac version without any errors. (I am running the develop branch)

jglezplatas, 2017-05-02:
Hi
I have tested in another MacOs and I have the same error.
Please, check it!

CPrescher, 2017-05-02:
Well, we check it and we know it is working, when you run it from python. But you are running the App... and did not tell us in your description, also which version of MacOS X you are using is missing. Sometimes it also good to just google the errors. This is an operation system specific error. 

Anyways, please see 3rd point in http://www.clemensprescher.com/programs/dioptas/dioptas-faq to solve your issue.

Best wishes,
Clemens

closed 2017-05-02 by CPrescher

jglezplatas, 2017-05-02:
> com/programs/dioptas/dioptas-faq to solve your issue.

Thanks

Javier

Saludos

    Javier Gonzalez-Platas

    *Departamento de Física *
*    Universidad de La Laguna*
*    Apdo. Correos 456*
*    E-38206 La Laguna, Tenerife*

    E-mail: jplatas@ull.edu.es
    Tfno:922318251