Dioptas/Dioptas #38: Request for angle based masking

CPrescher, 2017-05-02:
A user asked for a masking method, where a mask is created based on a cutoff 2-theta, q or d-spacing.

erangre, 2017-05-02:
Would having a separate toggle for the x-range from the pattern background answer this?

CJ-Wright, 2017-05-13:
You might also consider using some of the auto-masking in scikit-beam. It is in a function named `binned_outlier`.

CPrescher, 2026-09-21:
Implemented in commit https://github.com/Dioptas/Dioptas/commit/fbd9e7405f00fa37ec2a4121896a04d3bee558fa, pushed to develop.

The Mask panel now includes Range controls directly below the intensity thresholds. Choose 2θ (degrees), q (Å⁻¹), or d-spacing (Å), enter minimum and maximum bounds, and use Inside or Outside to select pixels. Inside includes both bounds; leaving either bound blank provides a single cutoff.

The actions use the current calibration, respect Mask/Unmask mode, preserve pixels outside the selection, and support undo/redo. Calibration is required, and q/d ranges also require a valid wavelength. The resulting pixel mask remains fixed if the calibration later changes.

Validation: 162 tests and 3 subtests passed across the focused masking, calibration, and undo/redo suite. Coverage includes range boundaries, open-ended cutoffs, all three units against independently calculated geometry, supersampling, configuration switching, and invalid input.

closed 2026-09-21 by CPrescher