Dioptas/Dioptas #45: Add images batch mode

erangre, 2017-06-12:
There is an issue when combining images in batch mode "Add", when the added intensity is beyond the limit for a single value in the TIFF file. This causes apparent "saturation" of the image and therefore the integrated intensities appear strange.
A possible solution is to perform an average instead of sum.
1. Will this create other issues (for instance, imprecise values when adding for a certain pixel multiple small values to a single relatively larger value)?
2. Should "add" be changed to "average" or should both options exist?

CPrescher, 2017-06-26:
fixed in 5c5ed06

closed 2017-06-26 by CPrescher

erangre, 2017-06-26:
Great.
If you save the added image as a TIF file, will it remain in 32 bit and is it still possible to open it in other software as a diffraction image?

CPrescher, 2017-06-26:
It is actually saved right now by default as int32 (not uint32 though). So if it is working right now, it should continue to work.  (see line 887 in ImageController.py)

erangre, 2017-06-26:
ok. so as long as someone doesn't add over ~2000 files we should be fine.

CPrescher, 2017-06-26:
which i think is rather unlikely, even then it would only matter if those files would have maximum intensity (where you probably do not need to add files anyways...)