Dioptas/Dioptas #69: Correcting .edf file with spline file

sedmak, 2018-04-11:
Hi, I'd like to correct the .edf files for a spatial distortion of the detector - .spline file. Is it possible to implement into Dioptas?

CPrescher, 2018-04-11:
Hi, in principle pyFAI support s spline files for spatial distoratio corrections, therefore it sould be possible to implement. However i have never played with it, do you have experience?

sedmak, 2018-04-13:
Hi, 
I managed to make it work with pyFAI with a few lines. 

import fabio
import pyFAI
from pyFAI.distortion import Distortion
import os,sys
from pyFAI import detectors

spline_file=('f4mnew.spline')  => spline file for our frelon detector
hd = pyFAI.detectors.FReLoN(splineFile=spline_file)
dis = Distortion(hd)#, resize=True)  => defines type of a distortion 

for i in(glob.glob('*.edf')):    => correct all the files in the current dir with .edf extension
    im=fabio.open(i)
    raw=im.data
    corrected=dis1.correct(raw1) 
    im.data=corrected
    im.write('cor_%s.edf'%i)

Is it possible to implement it somehow into Dioptas then? 
Thanks!

CPrescher, 2018-04-14:
Hi again,

would it be possible that you can upload a sample image, calibration and spline file, so that i can test that everything works correctly?

Best wishes,
Clemens

sedmak, 2018-04-16:
Hi, 
So here is an edf image of a CeO2 that we use for calibration and a spline file of the detector. 
Best,
Pavel
[files.zip](https://github.com/Dioptas/Dioptas/files/1915738/files.zip)

CPrescher, 2019-01-15:
So this is implemented now in the develop branch and will be in the next release, thank you for the test files.

closed 2019-01-15 by CPrescher