Dioptas/Dioptas #71: Error while loading file

baptiste7590, 2018-04-30:
Hi,

This error message appears when I try to load a file :

[image]

What should I do ?

Best wishes,
Baptiste

CPrescher, 2018-04-30:
Hi,

I do not know.
what file is it? Where do you got the image from, and could you maybe attach a sample image here in the comment?

Best wishes,
Clemens

baptiste7590, 2018-04-30:
It's a .tif file. 

I first treat a raw image (in .tif) with ImageJ and then when I save it (in .tif) I can't open it anymore with dioptas. However I can open a raw image without any problem.
I know that this treatment process works with Mac but not with Windows.

I can't attach an image here the .tif file is not supported.

CPrescher, 2018-04-30:
Why do you need to treat it with ImageJ? Is there a specific setting in ImageJ  you are using when saving it?

Could you please upload the original file (before imageJ), the file after treatment with imageJ and tell me what exactly you do with imageJ, so that i can do it myself?

Files can be uploaded packed in a zip file.

Best wishes,
Clemens

baptiste7590, 2018-05-09:
Hi Clemens,

Sorry for the late answer but I finally find a way to open the file.
After the treatment on ImageJ I just have to export the image instead of saving it under a certain file and I can open it on dioptas without any problem.

Thanks for your time.

Best wishes,
Baptiste

CPrescher, 2018-05-10:
Ok i am happy that the issue is solved for you. But I still would like to know what ImageJ does, that Dioptas does not like under windows....

closed 2018-05-10 by CPrescher