Dioptas/Dioptas #92: [FEATURE] Add ability to read multiple frame spe files.

erangre, 2019-03-28:
Currently when Dioptas reads an spe file with multiple frames it sends a list of images to Dioptas, which cannot handle it.

This issue was raised by an APS beamline which saves time resolved diffraction data in a single spe file.

Pull request #91 uses a temporary fix for this issue by choosing only the first frame (index 0).

Perhaps in the image window there should be scrolling between frames of the image. There might be other file formats which support multiple frames per image file which this would be useful for.

Treeed, 2019-04-20:
92b131e953a52ef9d575b1cc58ea9f14f78256c8 in #95 Adds support for image series. You could either look at the changes and implement it yourself or provide some sample data.
To activate the series mode load_spe would need to set "series_max" and "series_get_image" in "load_spe", similar to "load_lambda".

CPrescher, 2023-09-29:
This is an old issue. But Eran do you have an example SPE file?

erangre, 2023-09-29:
> This is an old issue. But Eran do you have an example SPE file?

I don't seem to have the files in question. But I think trying with any SPE file with multiple frames should give the same result.