Dioptas/Dioptas #97: Dioptas crashes when an image which only contains extreme values is loaded

Treeed, 2019-05-31:
When loading an image that contains only very high and very low values without any intermediates dioptas crashes. The bug can be reproduced with the attached file (zipped because github doesn't like tifs): [l1_new2_ramp1_00001_m1.zip](https://github.com/Dioptas/Dioptas/files/3242879/l1_new2_ramp1_00001_m1.zip)
The reason for the crash is that the histogram shown on the side of the image gets put through a logarithm twice - once in image changed: https://github.com/Dioptas/Dioptas/blob/072ea6209c9ef084ba6abc617f8a5d366ccdefca/dioptas/widgets/plot_widgets/HistogramLUTItem.py#L256-L257 and once in the widget itself, enabled here: https://github.com/Dioptas/Dioptas/blob/072ea6209c9ef084ba6abc617f8a5d366ccdefca/dioptas/widgets/plot_widgets/HistogramLUTItem.py#L116-L119 
after this the histogram only consists of -inf, inf and NaN. Pyqtgraph masks all of these values out and then tries to get the minimum and maximum of the now empty array, causing an error.

Treeed, 2019-05-31:
A possible fix would be to disable the internal logarithm of the widget, giving dioptas full control over the shown values. Because only one logarithm already has a much lower chance of producing infinites and NaNs this already fixes the issue for most images. 
Additionally the values can now be sanity checked and fixed before being passed to pyqtgraph if required.

erangre, 2019-05-31:
I have encountered this problem before, but somehow the attached file is not raising this error for me. I am using the current develop branch.

My environment:
Python 3.6.6.final.0
Pyqtgraph version 0.10.0 from channel cprescher
Numpy 1.15.4

Treeed, 2019-06-01:
It turns out the issue is indeed fixed in a newer pyqtgraph version. I assumed the version on conda-forge and pypi to be fairly recent but despite showing as the same version number when installed it is about a year older than the package from @CPrescher and almost 3 years out of date. 
The weird double logarithm remains but is more of a cosmetical issue.
Thank you for testing @erangre!

closed 2021-09-26 by CPrescher