Dioptas/Dioptas #105: GUI scaling issues on high DPI displays

pauliacomi, 2019-09-16:
On Windows machines with high DPI displays (4K), the GUI buttons are very small due to scalling issues, making it difficult to use the program.

A possible solution, which worked on my particular machine, was to allow Qt to perform automatic scaling by setting the following application attribute in `__init__.py`:

```
from pyqt import QtCore
QtWidgets.QApplication.setAttribute(QtCore.Qt.AA_EnableHighDpiScaling, True)

def main():
    ...
```

However, I am not sure how this setting would interact with other hardware and screen configurations.

erangre, 2019-09-17:
I have noticed with 4K displays that there is also a problem with the axes of the graphs. Did you also encounter this issue, and does this change you suggest fix that as well?

pauliacomi, 2019-09-17:
Initially I was going to say I don't observe issues with the axes. However, I did a little bit of digging.

When I switch the program from a 4K screen to a 1080p one, the following pattern scaling occurs in the graphs:

[image]

After the change I mentioned in the previous comment, this behaviour dissapears and the scaling looks identical between the two screens.

Again I should stress that this is on one single machine, so it would need a bit of testing. I would be happy to submit a pull request so that someone else could also have a look on their setup.

erangre, 2019-09-17:
Yes, this strange behavior of the axes is what I observed before on someone else's system.
I think that on one of my computers I can change the resolution from 1080p to 4K for both screens together and/or separately, so maybe I can test this.

pauliacomi, 2019-09-17:
Okay then, I went ahead with the PR.
Hopefully this could be a simple fix, otherwise more granular layout changes might be needed.

CPrescher, 2019-09-17:
Great thanks for a way of dealing with high dpi. I was always under the impression that QT does not support it..

I will test it on a couple of machines and wait for Erans tests.

Then we can merge.

pauliacomi, 2019-09-17:
To my knowledge, this setting was introduced recently (Qt 5.6) as an attempt to quickly allow apps to be compatible with high resolutions. It is new to me as well: I am not well aquainted with Qt so am unaware of its limitations.

Hope it works out!

erangre, 2019-09-25:
I tried to test this on Windows 8.1 but didn't encounter the problem at all (before the fix).
@CharitonS told me she tested it on two machines and encountered the problem one one of them and that the fix works for it. From the details she provided it seems that this issue is caused by Windows reporting the wrong resolution in some cases (but in some cases even with high DPI it reports the correct resolution and there is no issue with the Qt).

borlongioffei, 2020-12-20:
I recently encountered a similar problem for the linux version in a high DPI display. The interface is messed up and too big: it is almost impossible to use the software. The solution by the way is simple, it is sufficient to disable the DPI scaling just launching  the executable with the following setting:

`env QT_AUTO_SCREEN_SCALE_FACTOR=0  /path/to/dioptas
`

Hope this helps.

[image]

CPrescher, 2021-09-26:
I know changed it so, that the high dpi scaling only is enabled on Windows and Mac. Therefore, it does not need to be disabled for linux.

closed 2021-09-26 by CPrescher

borlongioffei, 2021-09-26:
Thank you for the information, Clemens.
You are building a great software and you are writing history in our field!
See you at the next EHPRG ;-)