Dioptas/Dioptas #108: Mac release doesn't install

aniryel, 2019-10-29:
I was trying to install the release for Mac, but after pressing "Open" nothing happens, the process disappears. Could you please help me out?

CPrescher, 2019-11-02:
Hi, 
could you please give me more information:

1. how did you install? (The executable, anaconda, ...?)
2. What OS X version are you using?
3. Did you try: FAQ number 3 in http://www.clemensprescher.com/programs/dioptas/dioptas-faq ?

aniryel, 2019-11-04:
Hi,

I have macOS Sierra, version 10.12.6, I was trying to install the executive for Mac you have on the Github with the last update.
I didn't try the FAQ number 3 as it doesn't give me any error. It just disappears from processes and nothing happens at all after I try to open the installation.

closed 2021-02-16 by CPrescher