Dioptas/Dioptas #113: Output y-error / XYE

PetMetz, 2020-05-02:
In the current version of Dioptas, the only integrated powder diffraction data format that outputs an error column for measured intensity is the GSAS / .fxye format. Can you please add another plain text data format including the y-error column (i.e.  .xye which is the standard data format for Topas).

Thanks!

CPrescher, 2020-05-03:
Could you attach a sample file for xye? Does it include a header? 

The more basic question for me, how much sense does it make to add errors to output files when they are only the sqrt of the intensity.  In my opinion it gives the false sense that the error is taken care off. However, it is clearly not.

pyFAI provides the possibility to get error estimates based on azimuthal variation. I think it might be nice to add this and be able to select.

PetMetz, 2020-05-03:
The .xye format used by Topas looks like the following

> ' comment lines begin with a single quote
> ' there is no limit on the number of comment lines
> ' there is no specific header for the following data
> ' which is white space delimited
> ' two-theta intensity error
> 0.001  10000.00000  0.12345678
> 0.002  20000.00000  0.12345678
> ...

If you wanted it to be less program-specific, a white-space-delimited ASCII data format with # comments or etc. would be equally helpful.

My assumption was that the error column in the GSAS data format was the azimuthal variation in the data. I agree this would be a nice feature to add.

erangre, 2020-05-03:
I think that I added the esd to the fxye output and I just followed an example of a fxye file I found, and I saw that there it was just the sqrt so I used that.

PetMetz, 2020-05-04:
The usual assumption is that X-ray counting is a random process involving where detection events are independent, and therefore that the intensity measured at a point in reciprocal space follows Poisson statistics. Thus, the naive approach to estimating uncertainty on powder diffraction intensities is to calculate sigma = sqrt(counts).

The 2D solid state detectors are potentially more problematic and more informative-- more problematic because X-ray detection is more convoluted than simply counting photons entering each pixel, and more informative because measuring the complete Debye ring is in effect performing N concurrent 1D powder diffraction experiments.

kif, 2022-01-06:
PyFAI provides the estimated error with 3 different approaches:
* The variance can be provided: if the detector's response is known, the variance can be propagated and provides the error as the sqrt of the summed intensity normalized by the contributing pixels.
* A special case of the former is assuming the detector to be Poissonian: `variance = max(1, intensity)`  for any pixel. 
* If one assumes no textures from the sample, the variance can be assessed from each ring (use `sigma-clipping` without clipping with an azimuthal error-model)