Dioptas/Dioptas #114: Bug when adding phases in .cif format

lgpmartins, 2020-05-13:
Hi,

I'm trying to add some phases in .cif format, since I've read that dioptas convert them to jcpds. However, I've received the following error message when I tried to load those files:

How to fix this?

Error information:

--------------------------------------------------------------------------------

2020-05-13, 09:33:55

--------------------------------------------------------------------------------

<class 'CifFile.StarFile.StarError'>:

Star Format error: Attempt to insert duplicate item name _chemical_formula_sum

--------------------------------------------------------------------------------

File "dioptas\controller\integration\PhaseController.py", line 134, in add_btn_click_callback

File "dioptas\controller\integration\PhaseController.py", line 150, in _add_phase

File "dioptas\model\PhaseModel.py", line 61, in add_cif

File "dioptas\model\util\cif.py", line 68, in convert_cif_to_jcpds

File "site-packages\CifFile\CifFile.py", line 2904, in ReadCif

File "site-packages\CifFile\StarFile.py", line 2065, in ReadStar

File "site-packages\CifFile\YappsStarParser_1_1.py", line 161, in input

File "site-packages\CifFile\YappsStarParser_1_1.py", line 181, in dblock

File "site-packages\CifFile\YappsStarParser_1_1.py", line 193, in dataseq

File "site-packages\CifFile\YappsStarParser_1_1.py", line 205, in data

File "site-packages\CifFile\StarFile.py", line 592, in AddItem

--------------------------------------------------------------------------------

Dioptas Version: 0.4.1

CPrescher, 2020-05-13:
Hi,

the error is written in the log. apparently your cif contains two times the item _chemical_formula_sum

please remove one of them by using a text editor. Also please try using a newer version of Dioptas.

Best wishes,
Clemens

CPrescher, 2020-05-21:
Does it work now?

lgpmartins, 2020-05-21:
Yes. It works.Thank you very much!

closed 2020-05-21 by lgpmartins