Dioptas/Dioptas #115: (New) Bug when adding phases in .cif format

lgpmartins, 2020-05-21:
Hi,

I've received the following error message when I tried to add a phase in .cif format:
I could not find what was not iterable in the cif.txt
How to fix this? 
I'm attaching the .cif file in txt.
[EntryWithCollCode202338.txt](https://github.com/Dioptas/Dioptas/files/4662615/EntryWithCollCode202338.txt)

Error information:
--------------------------------------------------------------------------------
2020-05-21, 09:14:49
--------------------------------------------------------------------------------
<class 'TypeError'>: 
argument of type 'NoneType' is not iterable
--------------------------------------------------------------------------------
  File "dioptas\controller\integration\PhaseController.py", line 134, in add_btn_click_callback
  File "dioptas\controller\integration\PhaseController.py", line 150, in _add_phase
  File "dioptas\model\PhaseModel.py", line 61, in add_cif
  File "dioptas\model\util\cif.py", line 69, in convert_cif_to_jcpds
  File "dioptas\model\util\cif.py", line 359, in __init__
  File "dioptas\model\util\cif.py", line 497, in get_symmetry_from_space_group_number
--------------------------------------------------------------------------------
Dioptas Version: 0.4.1

CPrescher, 2020-05-21:
Without seeing the cif file i cannot understand what is the issue here. But I suspect, that the cif file is not providing the Space group number.

Also please use the latest Version of Dioptas from  https://github.com/Dioptas/Dioptas/releases. Maybe your bug has already been fixed.

erangre, 2020-05-21:
It appears that the key is different from what the cif reader is looking for.

This is what the cif reader is looking for:
_symmetry_Int_Tables_number

This is what is in the file:
_space_group_IT_number

The H-M one is also different from what the cif reader is looking for.

lgpmartins, 2020-05-21:
> It appears that the key is different from what the cif reader is looking for.
> 
> This is what the cif reader is looking for:
> _symmetry_Int_Tables_number
> 
> This is what is in the file:
> _space_group_IT_number
> 
> The H-M one is also different from what the cif reader is looking for.

-------------------------------------------------------------------------

Hi, Thanks for the reply.
The issue is that I have other similar .cif files that also have this line ''space_group_IT_number'', such as the one I've attached, and they show no problem.

lgpmartins, 2020-05-21:
?Hi,

I cannot open the version 5.1 for some reason. I tried the 5.0 and ti works. Would you have an idea why?

erangre, 2020-05-21:
> > It appears that the key is different from what the cif reader is looking for.
> > This is what the cif reader is looking for:
> > _symmetry_Int_Tables_number
> > This is what is in the file:
> > _space_group_IT_number
> > The H-M one is also different from what the cif reader is looking for.
> 
> Hi, Thanks for the reply.
> The issue is that I have other similar .cif files that also have this line ''space_group_IT_number'', such as the one I've attached, and they show no problem.
> [ReO7 15217.txt](https://github.com/Dioptas/Dioptas/files/4663056/ReO7.15217.txt)
In the ReO7 you attached it doesn't find the information it is looking for under the tags it is familiar with, but it manages to find out that it is orthorhombic from the unit-cell parameters.
In the first file it has to know something about the space group to know if it is hexagonal or trigonal so it fails.

closed 2021-09-26 by CPrescher