Dioptas/Dioptas #116: Cannot open Dioptas 0.5.1 executable

lgpmartins, 2020-05-22:
Hi,

I have been trying to install the Dioptas 0.5.1 executable version for Windows 64 and it does not work. No error message shows up. Once I cick to open the Dioptas executable, nothing happens.

There was no problem when opening the executables of Dioptas 0.5.0 and 0.041. 

Could you help me with this?

CPrescher, 2020-05-23:
I just tried out whether Dioptas 0.5.1 runs on a new computer with a brand new windows installation and no libraries installed. It works without any issues. The only thing I need to do is to right click the Dioptas.exe and disable the security check under properties.

anyaL1966, 2020-06-02:
Hello, my system is Windows10, when trying to start Dioptas_Win64_0.4.1., error message appears: Error loading Python DLL: ...\python35.dll (error code 126). But this file (python35.dll) is contained in the program folder!  thanks for your help
[image]

CPrescher, 2020-06-02:
Hi,

please try extracting the zip to a folder first, and then run the Dioptas.exe file. Please also use the latest Version of Dioptas from: https://github.com/Dioptas/Dioptas/releases

Best wishes,
Clemens

closed 2021-02-16 by CPrescher