Dioptas/Dioptas #117: Can't start Dioptas 0.5 and 0.5.1 on Linux

catfluoride, 2020-05-28:
Hi! Trying to start Dioptas on Linux gives plenty of warnings/errors about Fontconfig and then fails with a segfault. I guess the problem may be bug or some odd configuration on my system (Arch Linux).

Messages are here: [https://pastebin.com/vb1WsEdy](https://pastebin.com/vb1WsEdy)

Any advice?
Thanks a lot in advance

CPrescher, 2020-05-28:
Hi!

I do unfortunately not have a clue about arch Linux. The linux packages are compiled for ubuntu/debian. I think in your case the best bet would be to use anaconda package. Just install anaconda python and then run 

conda install dioptas -c cprescher

dioptas can then be started by typing
dioptas

in the command line

catfluoride, 2020-05-28:
I'll try that. Thanks a lot for your kind suggestion!

closed 2020-05-28 by catfluoride