Dioptas/Dioptas #119: Reading images from ESRF in hdf5 format

jonwright, 2021-01-28:
Can you read data from ESRF in the new file formats? I tried and it seems to recognise h5 as an extension but I didn't see how to do it. I can do this and I see you use fabio:
```
>>> fabio.open("CeO2_0001.h5::/1.1/measurement/eiger")
<fabio.hdf5image.Hdf5Image object at 0x7f5992334450>
```
...but I did not see how to give this string via the file-open selector thing ?
Thanks for your help!

CPrescher, 2021-01-28:
Hi Jon,

could you upload an example file? Is the "::/1.1/measurement/eiger" part of the filename? 
I think the file selector wants a valid file and therefore would probably not work... Also I guess there has to be another way to open this file without a specification before loading...

jonwright, 2021-01-28:
The "::" is the separator to give the address inside the hdf5 file. The "/1.1/measurement/eiger" is the address of a hdf5 dataset inside the file. In most cases the frames are spread out in a series of other hdf5 files and this thing is a virtual dataset. There is a useful concept of a dataurl with the "silx" library here, http://www.silx.org/doc/silx/latest/modules/io/url.html , which leaves things open to having data in a cloud, etc.

I am happy to upload files somewhere but I need to dig around to find a small one, or generate something new. The default places many frames in each file, so they are usually pretty big.

jonwright, 2021-01-29:
Here is one with some blank pictures. There are two "scans" that both belong to the same dataset. In real experiments people are getting index level hdf files per experiment, per sample, and per dataset. To get monitor values and motor positions you need to look in one of the files above the images in the tree (if that makes sense).
[demoh5.zip](https://github.com/Dioptas/Dioptas/files/5893160/demoh5.zip)

CPrescher, 2021-04-30:
Hi Jon,

i am currently trying to implement this into Dioptas. I would like to have the user be able to choose from selected data sources (e.g. parse through the hdf5 file, identify image data (shape length==3) and then make these choices available via a combobox).

I found that the data is twice linked inside of the file.
e.g.:
/demoh5_dataset_1.1/instrument/marana/data/
/demoh5_dataset_1.1/measurement/marana

have the same data. Why is this? Will this be for all new ESRF hdf5 structures like this?

Further, I was wondering why the datasets of individual files e.g. in scan0001/marana_0000.h5 have a shape of (100,:, :), even though only 2 images are saved within...

jonwright, 2021-04-30:
Hi Clemens,

I am not the expert on this - maybe we need to get an expert from the data analysis unit. From what I understand :

The data were supposed to follow the nexus standard. The "measurement" entries are usually things which are changing during a scan and the "instrument" entries represent some kind of overview of the whole experiment. I think the measurement group is often a lot of soft links to the instrument group.

There should be nexus tags like attributes NXdetector to identify things. You might get a single 2D frame in some cases, instead of 3D stack.

For the files which are padded: they are written by the Lima device servers and the 100 was the frames-per-file blocksize that was set by the user. Then only two frames were actually written.

I don't know how this compares to hdf5 data coming out of other synchrotrons, but it is in "production" for about a year now and has not changed significantly (unless you are decoding the VDS links).

The silx code might be useful:
http://www.silx.org/doc/silx/latest/modules/gui/dialog/groupdialog.html#module-silx.gui.dialog.GroupDialog

Thanks!!! Our users will be very happy

CPrescher, 2021-05-03:
Hi Jon,

thank you. I am a little confused about that is supposed to be following the Nexus standard. Usually, this would mean that groups are named with NX prefix, like NXdetector etc. Which is not followed within these files. Nexus also uses NXentry as 
 individual measurements and has saved all the parameters for this measurement inside of this entry. The ESRF files clearly deviate from this structure.

A good thing would also a tag to identify whether data is actually image data, then crawling through the file would be easier...

I have it working now, but i will leave the duplicate entries selectable and hope this is not too bad. How many measurement series do you usually have in one file?

Best wishes,
Clemens

jonwright, 2021-05-03:
Hi Clemens

I thought that "NXdetector" is a group attribute? You name the groups what you like because most people have several detectors with names like "beamviewer" or "eiger" and they use these for the group names. The nexus tags are then in attributes. I think Armando got the nexus standard changed to allow "1.1" as a group name, so the files should be compliant now.

Selectable duplicates seems like the best option to me. People may also want to come in via a "technique" group structure in the future.

For sizes, it ranges from 1 frame to >1 million frames for XRD-CT or imaging. More and more now, people are scanning the sample in the beam to find a region of interest. If we run at 500 fps then 33 minutes is 1e6 frames, you don't even have time to eat a good lunch!

Thanks!

Jon

CPrescher, 2021-05-03:
Hi Jon,

I am sorry, you are absolutely correct. The NX classes are saved in the atts['NX_class']. I thought i have seen this used as actual names in other nexus files previously.

I can imagine with everything being that fast now, there is a lot of time pressure.

I was also wondering on not how many images you collect per measurement, but how many measurement series you usually save in one parent hdf5 file (e.g.  /demoh5_dataset_1.1/measurement/marana, /demoh5_dataset_1.2/measurement/marana and so on).

If this is more than 20 or 30. A combobox might be annoying to scroll though all those measurements.

Best wishes,
Clemens

jonwright, 2021-05-03:
Hi Clemens, 
The "top level" experiment file can hold everything we do during a 6 day experiment. In this case people would do better to open at the "sample" or "dataset" level. There is a tree of experiment/beamline/sample/dataset/scan in the file system, so I guess people can work around it. 
Thanks!
Jon

commit 10601f74 (2021-05-04): add capability for reading hdf5 files from ESRF, closes #119
closed 2021-05-04 by CPrescher