Dioptas/Dioptas #122: calibration error

zjwdicp, 2021-08-16:
When load the image for calibration
An unhandled exception occurred. Please report the bug under: https://github. com/Dioptas/Dioptas/issues or via email to:
<clemens. prescher@ gnail. com>.
Please make sure to report the steps to reproduce the error. Otherwise it will be hard to fix it.
A log has been written to "D:\ Dioptas_Win64_0.5.2\ Dioptas_Win64_0.5.2\ dioptas\ error. log".
Error information:
2021-08-16,10:35:33
Kclass' AssertionError': 
File "dioptastcontroller\ CalibrationController. py", line 465, in calibrate 
File"dioptas\ model\ Calibratiornllodel. py", line 301, in calibrate 
File "pyFAT\ geometryRefinement. py", line 104, in _init Dioptas Version:0.5.2

CPrescher, 2021-08-23:
how did you load the image? By clicking load calibration? The image should be loaded by clicking load image on the upper right.

zjwdicp, 2021-08-24:
I loaded from the image you pointed， I also copy the program to another computer of Win10， the same problem continue，I do not know how to fix this， thank you！

CPrescher, 2021-08-24:
Did you select initial points before calibration, e.g. on the first ring? Dioptas needs an initial guess (described in manual)

closed 2021-09-26 by CPrescher