Dioptas/Dioptas #133: BUG Debian unstable

farhi, 2022-11-18:
Hello Clemens,

we have pushed a Debian package for Dioptas:
- https://tracker.debian.org/pkg/dioptas

It provides an easy installation for e.g. Debian., Ubuntu, Mint, etc. However, in its development version with python 3.10 and Qt 5.15, there is an error showing-up at start:
```
An unhandled exception occurred. Please report the bug under:
https://github.com/Dioptas/Dioptas/issues
or via email to:
<clemens.prescher@gmail.com>.

Please make sure to report the steps to reproduce the error. Otherwise it will be hard to fix it.
A log has been written to "/usr/lib/python3/dist-packages/dioptas/error.log".

Error information:
--------------------------------------------------------------------------------
2022-11-18, 15:56:34
--------------------------------------------------------------------------------
<class 'ValueError'>:
Unable to convert ffffff to QColor

--------------------------------------------------------------------------------
File "/usr/bin/dioptas", line 9, in <module>
main()
File "/usr/lib/python3/dist-packages/dioptas/__init__.py", line 109, in main
controller = MainController()
File "/usr/lib/python3/dist-packages/dioptas/controller/MainController.py", line 48, in __init__
self.widget = MainWidget()
File "/usr/lib/python3/dist-packages/dioptas/widgets/MainWidget.py", line 113, in __init__
self.calibration_widget = CalibrationWidget(self)
File "/usr/lib/python3/dist-packages/dioptas/widgets/CalibrationWidget.py", line 46, in __init__
self.calibration_display_widget = CalibrationDisplayWidget(self)
File "/usr/lib/python3/dist-packages/dioptas/widgets/CalibrationWidget.py", line 323, in __init__
self.pattern_widget = PatternWidget(self.pattern_layout_widget)
File "/usr/lib/python3/dist-packages/dioptas/widgets/plot_widgets/PatternWidget.py", line 43, in __init__
self.create_main_plot()
File "/usr/lib/python3/dist-packages/dioptas/widgets/plot_widgets/PatternWidget.py", line 74, in create_main_plot
self.legend.addItem(self.plot_item, '')
File "/usr/lib/python3/dist-packages/dioptas/widgets/plot_widgets/ExLegendItem.py", line 134, in addItem
label.setText(name)
File "/usr/lib/python3/dist-packages/pyqtgraph/graphicsItems/LabelItem.py", line 57, in setText
color = fn.mkColor(color)
File "/usr/lib/python3/dist-packages/pyqtgraph/functions.py", line 272, in mkColor
raise ValueError(f"Unable to convert {c} to QColor")
--------------------------------------------------------------------------------
Dioptas Version: 0.5.2
```

Some python 3.10 code now request variable types (and avoid casting). It seems to be the case with Qt5.

Cheers, Emmanuel.

farhi, 2022-11-18:
This is pyqtgraph 0.13.1
- https://packages.debian.org/sid/python3-pyqtgraph

farhi, 2022-11-18:
You are right. A Ubuntu 22.04 with dioptas + python3-epics runs perfectly. I'll inquire the library versions, to spot which one is different between the 22.04 and the current Debian unstable.

farhi, 2022-11-21:
I have gone through the list of packages you gave above, and upgraded them one by one using `pip3 install`. However, this did not solve the issue for Dioptas 0.5.2

Starting from a Debian unstable (18 Nov 2022), I can have Dioptas work from the git repo with:
- git clone https://github.com/Dioptas/Dioptas.git
- cd Dioptas
- pip3 install .
Successfully installed altgraph-0.17.3 dioptas-0.5.5.dev59+g444b7864 hdf5plugin-3.3.1 pyinstaller-5.6.2 pyinstaller-hooks-contrib-2022.13 sharedmem-0.3.8

farhi, 2022-11-21:
Running:
- `apt install python3-altgraph` 
- `pip3 install hdf5plugin`

alone do not allow to run Dioptas 0.5.2. So the conclusion is that upgrading Dioptas is required. The latest official release 0.5.4 does not run either.

Using the current git repo solves the issue on Debian unstable/sid (12)
- git clone https://github.com/Dioptas/Dioptas.git
- cd Dioptas
- pip3 install .

Do you plan to release a new version soon (0.5.5) ?

CPrescher, 2022-11-21:
Hi, 
yes I should do a new release 0.5.5 within the next 2 weeks. Would this be fast enough for you?

farhi, 2022-11-21:
Yes this is good timing. We shall then push the new upstream code from Dioptas git into the Debian package.
Thanks, Emmanuel.

CPrescher, 2023-09-29:
since we are already on 0.5.8, I think this can be closed

closed 2023-09-29 by CPrescher