Dioptas/Dioptas #134: Could you integrate  the Debian patches into dioptas ?

picca, 2022-11-22:
Hello, here you can find a bunch of patch that we need to apply in order to build the Debian package of dioptas.

https://salsa.debian.org/python-team/packages/dioptas/-/tree/debian/master/debian/patches

Could you evaluate if you could integrate the 1, 2 and 4.

the 5th is maybe already resolved.

thanks for considering

Frederic

CPrescher, 2022-11-22:
Hi Frederic,

i just pushed a new release, i think 1 and 2 should be fixed. Not sure, whether i like 4, have to think about it.

Best wishes,
Clemens

picca, 2022-11-22:
> Hi Frederic,

Hello Clemens,

> i just pushed a new release, i think 1 and 2 should be fixed.

thanks

> Not sure, whether i like 4, have to think about it.

:))

Cheers

Fred

picca, 2022-11-22:
for 4 I got rid of it.

thanks

closed 2022-11-22 by picca