Dioptas/Dioptas #135: OSError: [Errno 24] inotify instance limit reached

picca, 2022-11-22:
Hello, while preparing dioptas 0.5.5, I get this error message.

does it rings something for you ?

thanks

Frederic

```
test_automatic_file_processing (dioptas.tests.controller_tests.test_ImageController.ImageControllerTest.test_automatic_file_processing) ... Traceback (most recent call last):
  File "/<<PKGBUILDDIR>>/.pybuild/cpython3_3.11_dioptas/build/dioptas/controller/integration/ImageController.py", line 931, in auto_process_cb_click
    self.model.img_model.autoprocess = self.widget.autoprocess_cb.isChecked()
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/<<PKGBUILDDIR>>/.pybuild/cpython3_3.11_dioptas/build/dioptas/model/ImgModel.py", line 815, in autoprocess
    self._directory_watcher.activate()
  File "/<<PKGBUILDDIR>>/.pybuild/cpython3_3.11_dioptas/build/dioptas/model/util/NewFileWatcher.py", line 94, in activate
    self._start_observing()
  File "/<<PKGBUILDDIR>>/.pybuild/cpython3_3.11_dioptas/build/dioptas/model/util/NewFileWatcher.py", line 104, in _start_observing
    self.observer.start()
  File "/usr/lib/python3/dist-packages/watchdog/observers/api.py", line 262, in start
    emitter.start()
  File "/usr/lib/python3/dist-packages/watchdog/utils/__init__.py", line 93, in start
    self.on_thread_start()
  File "/usr/lib/python3/dist-packages/watchdog/observers/inotify.py", line 118, in on_thread_start
    self._inotify = InotifyBuffer(path, self.watch.is_recursive)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/watchdog/observers/inotify_buffer.py", line 35, in __init__
    self._inotify = Inotify(path, recursive)
                    ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/watchdog/observers/inotify_c.py", line 155, in __init__
    Inotify._raise_error()
  File "/usr/lib/python3/dist-packages/watchdog/observers/inotify_c.py", line 400, in _raise_error
    raise OSError(errno.EMFILE, "inotify instance limit reached")
OSError: [Errno 24] inotify instance limit reached
ERROR
```

CPrescher, 2022-11-22:
Hi,

i do not have this error neither on windows nor on linux.  but i am only on python 3.10. But I doubt python 3.11 causing this

CPrescher, 2023-09-29:
Closing, due to no new activity.

closed 2023-09-29 by CPrescher