Dioptas/Dioptas #138: Loading Phases Out

jlgonzal12, 2022-12-05:
When loading cif files into Dioptas to compare phases they do not load. 
[image]

CPrescher, 2022-12-06:
Hi,

there could be several reasons for this. What CIF file are you loading? Is it a "good" cif file? Where did you get it from? Can you check it under https://checkcif.iucr.org/ ?

Please also use the latest Dioptas version which is 0.5.5. you can download it from https://github.com/Dioptas/Dioptas/releases/tag/0.5.5

0.5.0 is from 2019 and I can for example see, that there was an bugfix related to the space group number for version 0.5.1:

loading cif files with the newer '_space_group_IT_number' representation work now correctly

Best wishes,
Clemens

CPrescher, 2023-04-24:
I will close this due to inactivity

closed 2023-04-24 by CPrescher