Dioptas/Dioptas #141: poni calibration file

rayguna, 2023-04-19:
I found that the poni calibration file generated using Dioptas does not seem to correspond to the poni calibration file generated using pyFAI. When I used Dioptas generated poni file to integrate my sample diffraction files, the output is incorrect. 

One obvious difference is that Dioptas generated poni file does not include the pixel size [2048, 2048]. Manually adding this information, however, yield the same output.

I notice that Dioptas has the option to toggle the calibration parameters between Fit2d Parameters, pyFAI parameters, and Calibration Parameters. When I view the pyFAI parameters, it clearly displays the parameters and the corresponding units, which agree with pyFAI. I also see that the same numbers that are displayed in pyFAI parameters are saved into the poni file. The first line in both of the Dioptas and pyFAI generated poni files show the same message "C-Order, 1 refers to the Y axis, 2 to the X axis". 

_How do I make the Dioptas poni file compatible with pyFAI and vice versa?_ 

Below are the two respective poni files generated from the same image.

**(1) Generated using pyFAI**
```
# Nota: C-Order, 1 refers to the Y axis, 2 to the X axis 
# Calibration done at Tue Apr 18 17:03:40 2023
poni_version: 2
Detector: Detector
Detector_config: {"pixel1": 7.9e-05, "pixel2": 7.9e-05, "max_shape": [2048, 2048]}
Distance: 0.09042654890599626
Poni1: 0.08770208056728616
Poni2: 0.08301875192228053
Rot1: 0.0008991170269026089
Rot2: 0.014297324876869731
Rot3: 8.410747801681629e-08
Wavelength: 5.2605892794704914e-11
```

**(2) Generated using Dioptas**
```
# Nota: C-Order, 1 refers to the Y axis, 2 to the X axis 
# Calibration done at Thu Mar  9 11:26:07 2023
poni_version: 2
Detector: Detector
Detector_config: {"pixel1": 7.9e-05, "pixel2": 7.9e-05, "max_shape": ""}
Distance: 0.09050119464827933
Poni1: 0.07472688167662014
Poni2: 0.0824074039494132
Rot1: -0.0060714861702728185
Rot2: -0.020617873558982407
Rot3: -1.0149682561991543e-09
Wavelength: 5.2589e-11
```

CPrescher, 2023-04-20:
Hi,

as I wanted to smooth out the transition from Fit2D, i made the choice to have the same image orientation as in Fit2D for Dioptas. pyFAI did not make this choice (even though pyFAI will give you fit2d calibration, it will not work within Fit2d due to this). Thus, to use the calibration generated within Dioptas with pyFAI or vice versa, you must invert the image vertically by doing:

image_array[::-1]

Best wishes,
Clemens

rayguna, 2023-04-20:
It works. I only had to invert the image, not the mask. Thanks.

closed 2023-04-24 by CPrescher