Dioptas/Dioptas #142: radial bins and output number of lines don't match

rayguna, 2023-04-21:
I am trying to save my image integration to a certain value by setting the radial bins parameter, but the output number of lines produced does not match. For instance, when I set the radial bins parameter to 2000, the output number of lines is 1674. When I set the radial bins parameters to 1000, the output number of lines 837. 

_Are there any other parameters that need to be set to get the number of lines in the output to match the set radial bins parameter?_

[image]

CPrescher, 2023-04-22:
This might be due to integrated intensities at certain positions being exactly zero. Zero values are removed in Dioptas. There is in principle no issues with full images and also for masked images it works correctly. Could you try masking your zero intensity areas on the detector? (e.g. by using below Threshold and set value to 1)

rayguna, 2023-04-24:
After applying a mask with a threshold that is set to below 1 and saving the data with a radial bins parameter of 2000, the generated number of lines did increase from 1674 to 1783, but not the full 2000.

CPrescher, 2023-04-24:
Would it be possible for u to share one Image and calibration to check it?

CPrescher, 2023-04-26:
a change to the most recent version of Dioptas, was fixing this issue.

closed 2023-04-26 by CPrescher