Dioptas/Dioptas #150: Implementing log scale display for images and integrated patterns

CPrescher, 2023-06-05:
Zhidan Zheng was suggesting to implement log scale for integrated patterns and images --> which apprently is the default mode with for SAXS data --> need to figure out, how to handle with <=0 data though

erangre, 2023-06-05:
How about when displaying in log that it shifts all the data by 1+max(0, -min_val) so that the smallest value would give 0 after log?

CPrescher, 2023-06-05:
i think, this is in principle a good option for displaying, although very often detectors extra have their masked pixels as (-1 or 0) - maybe this should be removed before then?
The other question is whether this changes anything for data analysis for SAXS -- I unfortunately have no experience with this.