Dioptas/Dioptas #151: Cannot open CBF files on Dioptas versions 0.5.3 to 0.5.7

isidrocg, 2023-06-06:
Since version 0.5.3 we are unable to load CBFs produced at XALOC Beamline (ALBA Synchrotron).
Here is the error information:
```
Error information:
--------------------------------------------------------------------------------
2023-06-06, 17:35:53
--------------------------------------------------------------------------------
<class 'IndexError'>: 
list index out of range
--------------------------------------------------------------------------------
  File "dioptas\controller\CalibrationController.py", line 184, in load_img
  File "dioptas\model\ImgModel.py", line 142, in load
  File "dioptas\model\ImgModel.py", line 167, in get_image_data
  File "dioptas\model\ImgModel.py", line 231, in load_fabio
  File "dioptas\model\loader\FabioLoader.py", line 31, in __init__
  File "fabio\openimage.py", line 180, in openimage
  File "fabio\cbfimage.py", line 236, in read
  File "fabio\cbfimage.py", line 119, in _readheader
  File "fabio\cbfimage.py", line 163, in _read_cif_header
  File "fabio\cbfimage.py", line 921, in __init__
  File "fabio\cbfimage.py", line 944, in _parse
--------------------------------------------------------------------------------
Dioptas Version: 0.5.7
```
After checking the error on Dioptas and the involved libraries, I found that there was a problem when parsing the header of the images. I suspect from the changes introduced in Dioptas version 0.5.3, which added support for reading ESRF hdf5 data files.
I inspected the headers generated at the beamline and determined that the problem is produced by "\\" escape characters, probably introduced by CBFlib during the generation of the images on long lines. All the other processing pipelines used at the beamline are parsing correctly the header.
Example of header giving errors:
```
###CBF: VERSION 1.7.11
# CBF file written by CBFlib v0.9.7

data_image_0

_Misc.acq_autoexpo_mode OFF
_Misc.acq_expo_time 0.04905
_Misc.acq_latency_time 0
_Misc.acq_mode Single
_Misc.acq_nb_frames 1
_Misc.acq_trigger_mode ExtTrigSingle
_Misc.image_bin <1x1>
_Misc.image_flip '<flip x : False,flip y : False>'
_Misc.image_roi <0,0>-<2463x2527>
_Misc.image_rotation Rotation_0

_array_data.header_convention PILATUS_1.2
_array_data.header_contents
;\

# Detector: PILATUS3X 6M, S/N 60-0140, XALOC, Alba
# 2023-06-01T11:23:57.246
# Pixel_size 172e-6 m x 172e-6 m
# Silicon sensor, thickness 0.001 m
# Oscillation_axis omega (X, CW)
# Chi 0 deg.
# Angle_increment 0.100000 deg.
# Polarization 0.99
# N_oscillations 1
# Beam_xy (1226.85, 1300.89) pixels
# Exposure_time 0.0491
# Phi 0.0000 deg.
# N_excluded_pixels = 1178
# Detector_distance 0.74362 m
# Detector_Voffset 0 m
# Alpha 0 deg.
# Threshold_setting 6334.000000 eV
# Exposure_period 0.0500
# Kappa -0.0004 deg.
# Start_angle -134.9998 deg.
# Detector_2theta 0.0000 deg
# Flux 6.843e+11
# Excluded_pixels :\
 /var/local/lib/dectris/config/calibration/Mask/badpix_mask.tif
# Count_cutoff 370913 counts
# Tau 1.991e-07 s
# Image_path :\
 /beamlines/bl13/projects/cycle2023-I/2023XXXXXX-xxxxxx/DDDDDDDD/RAW_DATA/B1X1
# Filter_transmission 0.2995
# Wavelength 0.978722973848
;
```
Removing manually the "\\" escape characters and the new lines (lines without "#") on the first line of the header and on the Image_path and Excluded_pixels lines allows Dioptas to load the images.

This error has been reproduced in versions 0.5.3 to 0.5.7, with the same image header  that was working with version 0.5.2.

We asked our users using Dioptas to reproduce this error and they told us that they are not updating to the last versions.

I can provide more info if required.

isidrocg, 2023-06-07:
Sure, thanks! 
[ref-test_5_0001.zip](https://github.com/Dioptas/Dioptas/files/11673815/ref-test_5_0001.zip)

CPrescher, 2023-09-29:
This should be fixed after a future release of fabio.

closed 2024-01-05 by CPrescher