Dioptas/Dioptas #156: Cannot integrade or calibrate

XRDEve, 2023-08-05:
An unhandled exception occurred. Please report the bug under:
 	https://github.com/Dioptas/Dioptas/issues
or via email to:
	 <clemens.prescher@gmail.com>.

Please make sure to report the steps to reproduce the error. Otherwise it will be hard to fix it. 

A log has been written to "C:\Users\evang/dioptas_error.log".

Error information:
--------------------------------------------------------------------------------
2023-08-05, 09:27:54
--------------------------------------------------------------------------------
<class 'AssertionError'>: 

--------------------------------------------------------------------------------
  File "dioptas\controller\CalibrationController.py", line 465, in calibrate
  File "dioptas\model\CalibrationModel.py", line 305, in calibrate
  File "pyFAI\geometryRefinement.py", line 105, in __init__
--------------------------------------------------------------------------------
Dioptas Version: 0.5.7

CPrescher, 2023-09-29:
Hi, already some time has passed, but could you solve your problem? I think the main issue here, is, that no ring was selected before you clicked calibrate.

CPrescher, 2023-11-13:
Closing due to inactivity

closed 2023-11-13 by CPrescher