Dioptas/Dioptas #160: Use the same colormap config for all tabs

t20100, 2023-09-05:
When switching tabs ("Calibration", "Mask", "Integration"), the zoomed area is preserved all the time. The colormap levels are only preserved for the first opening (fixed in PR #159) and the color LUT is never preserved. Newer colormap configurations like normalization will not be preserved as well.

I would propose to preserve the colormap config (levels, color LUT, normalization,...) when switching from one tab to the next for every switch (not just the first one) as it is done for the zoomed area.

Related to PR #159

CPrescher, 2023-09-12:
I was thinking about this multiple times, but I would preserve them to be different, since each tab has a different purpose. One could also argue about having the same focused center point and zoom level, but from a workflow perspective this is not necessarily good.
It is important that the calibration and mask img widgets autoscale when loading an image. In the current state they are not doing this.

t20100, 2023-09-12:
> It is important that the calibration and mask img widgets autoscale when loading an image. In the current state they are not doing this.

Also mentionned in https://github.com/Dioptas/Dioptas/pull/162#issuecomment-1710330717

This is due to PR https://github.com/Dioptas/Dioptas/pull/159, let's revert that for now then.

t20100, 2023-09-12:
> I would preserve them to be different, since each tab has a different purpose.

Yet they all show the same image, but indeed you might want to tweak the colormap differently for each step.

t20100, 2023-09-12:
At the same time, once you've set-up a colormap you might want to keep it through

CPrescher, 2023-11-13:
this should be done now

closed 2023-11-13 by CPrescher