Dioptas/Dioptas #173: Cannot install Dioptas==0.5.8 with pip

t20100, 2023-11-08:
Hi,

Running `pip install dioptas` to install v0.5.8 fails with `RuntimeError: This does not appear to be a Git project`.

`pip install dioptas==0.5.7` works fine.

<details><summary>Full message</summary>

```
Collecting dioptas
  Downloading dioptas-0.5.8.tar.gz (1.1 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.1/1.1 MB 6.2 MB/s eta 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [26 lines of output]
      Traceback (most recent call last):
        File "python3.10/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
        File "python3.10/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
        File "python3.10/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 149, in prepare_metadata_for_build_wheel
          return hook(metadata_directory, config_settings)
        File "/tmp/pip-build-env-kgq1ejrg/overlay/lib/python3.10/site-packages/poetry/core/masonry/api.py", line 42, in prepare_metadata_for_build_wheel
          poetry = Factory().create_poetry(Path(".").resolve(), with_groups=False)
        File "/tmp/pip-build-env-kgq1ejrg/overlay/lib/python3.10/site-packages/poetry_dynamic_versioning/patch.py", line 26, in alt_poetry_create
          name = _get_and_apply_version(
        File "/tmp/pip-build-env-kgq1ejrg/overlay/lib/python3.10/site-packages/poetry_dynamic_versioning/__init__.py", line 485, in _get_and_apply_version
          version = _get_version(config, name)
        File "/tmp/pip-build-env-kgq1ejrg/overlay/lib/python3.10/site-packages/poetry_dynamic_versioning/__init__.py", line 312, in _get_version
          version = _get_version_from_dunamai(vcs, pattern, config)
        File "/tmp/pip-build-env-kgq1ejrg/overlay/lib/python3.10/site-packages/poetry_dynamic_versioning/__init__.py", line 264, in _get_version_from_dunamai
          return Version.from_vcs(
        File "/tmp/pip-build-env-kgq1ejrg/overlay/lib/python3.10/site-packages/dunamai/__init__.py", line 1934, in from_vcs
          return cls._do_vcs_callback(
        File "/tmp/pip-build-env-kgq1ejrg/overlay/lib/python3.10/site-packages/dunamai/__init__.py", line 1973, in _do_vcs_callback
          return callback(**kwargs)
        File "/tmp/pip-build-env-kgq1ejrg/overlay/lib/python3.10/site-packages/dunamai/__init__.py", line 1066, in from_git
          _detect_vcs(vcs, path)
        File "/tmp/pip-build-env-kgq1ejrg/overlay/lib/python3.10/site-packages/dunamai/__init__.py", line 355, in _detect_vcs
          raise RuntimeError(
      RuntimeError: This does not appear to be a Git project
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
```

</details>

CPrescher, 2023-11-08:
this seems to be related to errors in the dunamai package?

t20100, 2023-11-08:
No sure, it seems to me `poetry_dynamic_versioning` is trying to get the version from `git` as in development mode.

CPrescher, 2023-11-08:
this might be true, since it is not a git repository when building from source it fails...
The windows install works, because it uses the wheel. I guess I need to create wheels for the other versions too...

t20100, 2023-11-08:
> I guess I need to create wheels for the other versions too

Eventually, but it is still expected that installation from pypi's tarball works. To me there should be a way with poetry to generate a sdist tarball which does not rely on git tags?

t20100, 2023-11-08:
I managed to build a tarball that can be installed with `pip` by: checking-out 0.5.8 tag, installing `poetry-dynamic-versioning` and running `poetry build --format=sdist`.

CPrescher, 2023-11-08:
Ok, good to know. It is impossible to overwrite something on pypi, that would mean, I create an updated version, e.g. 0.5.9 and upload the build with the command you mentioned

CPrescher, 2023-11-08:
Ok, good to know. It is impossible to overwrite something on pypi, that would mean, I create a new version, e.g. 0.5.9 and upload the build with the command you mentioned

CPrescher, 2023-11-13:
i made a new release, which creates wheels for every python version and every operating system. Can you try if it works now for 0.5.9 for you?

t20100, 2023-11-14:
Hi,

In a fresh virtual environment, installing Dioptas (itself) from source is fixed: `pip install --no-deps --no-cache dioptas==0.5.9 --no-binary dioptas` works while it didn't for v0.5.8, thanks!

However the new dependency `xypattern` cannot be installed from source: same issue as dioptas 0.5.8: `pip install --no-deps --no-cache xypattern --no-binary xypattern` leads to `RuntimeError: This does not appear to be a Git project`.

<details><summary>Logs</summary>

```
Collecting xypattern
  Downloading xypattern-1.0.4.tar.gz (11 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [26 lines of output]
      Traceback (most recent call last):
        File "testenv/lib/python3.10/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
        File "testenv/lib/python3.10/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
        File "testenv/lib/python3.10/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 149, in prepare_metadata_for_build_wheel
          return hook(metadata_directory, config_settings)
        File "/tmp/pip-build-env-i76nvbba/overlay/lib/python3.10/site-packages/poetry/core/masonry/api.py", line 42, in prepare_metadata_for_build_wheel
          poetry = Factory().create_poetry(Path(".").resolve(), with_groups=False)
        File "/tmp/pip-build-env-i76nvbba/overlay/lib/python3.10/site-packages/poetry_dynamic_versioning/patch.py", line 26, in alt_poetry_create
          name = _get_and_apply_version(
        File "/tmp/pip-build-env-i76nvbba/overlay/lib/python3.10/site-packages/poetry_dynamic_versioning/__init__.py", line 485, in _get_and_apply_version
          version = _get_version(config, name)
        File "/tmp/pip-build-env-i76nvbba/overlay/lib/python3.10/site-packages/poetry_dynamic_versioning/__init__.py", line 312, in _get_version
          version = _get_version_from_dunamai(vcs, pattern, config)
        File "/tmp/pip-build-env-i76nvbba/overlay/lib/python3.10/site-packages/poetry_dynamic_versioning/__init__.py", line 264, in _get_version_from_dunamai
          return Version.from_vcs(
        File "/tmp/pip-build-env-i76nvbba/overlay/lib/python3.10/site-packages/dunamai/__init__.py", line 1934, in from_vcs
          return cls._do_vcs_callback(
        File "/tmp/pip-build-env-i76nvbba/overlay/lib/python3.10/site-packages/dunamai/__init__.py", line 1973, in _do_vcs_callback
          return callback(**kwargs)
        File "/tmp/pip-build-env-i76nvbba/overlay/lib/python3.10/site-packages/dunamai/__init__.py", line 1066, in from_git
          _detect_vcs(vcs, path)
        File "/tmp/pip-build-env-i76nvbba/overlay/lib/python3.10/site-packages/dunamai/__init__.py", line 355, in _detect_vcs
          raise RuntimeError(
      RuntimeError: This does not appear to be a Git project
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
```

</details>

Also, the wheels do install on OS older than ubuntu 22.04 (it's the case here...) and then the source distribution is used.

CPrescher, 2023-11-14:
Ok, i fixed the xypattern issue. I actually accidentally put this dependency in this release, it is currently not used yet in dioptas, but it should replace the util/pattern in the future - I use this class on several projects, and thought it is better to have one simple package for this.

t20100, 2023-11-15:
Thanks!
I proposed PR #176 to fix the last small blocking point preventing an installation from source.

PR #176 (2024-02-06): Update xypattern version pinning
This PR relaxes the version of xypattern to use so `1.0.5` can be used which allows to install from source distribution

closes #173
closed 2024-02-06 by CPrescher