Dioptas/Dioptas #183: Error while loading file

rorajo195, 2024-01-31:
Hi,

This error message appears when I try to load a file. 

Error information:
--------------------------------------------------------------------------------
2024-01-31, 17:13:36
--------------------------------------------------------------------------------
<class 'OSError'>: 
Unable to open file (file signature not found)
--------------------------------------------------------------------------------
  File "dioptas\controller\CalibrationController.py", line 184, in load_img
  File "dioptas\model\ImgModel.py", line 142, in load
  File "dioptas\model\ImgModel.py", line 167, in get_image_data
  File "dioptas\model\ImgModel.py", line 286, in load_hdf5
  File "dioptas\model\loader\hdf5Loader.py", line 32, in __init__
  File "h5py\_hl\files.py", line 567, in __init__
  File "h5py\_hl\files.py", line 231, in make_fid
  File "h5py\_objects.pyx", line 54, in h5py._objects.with_phil.wrapper
  File "h5py\_objects.pyx", line 55, in h5py._objects.with_phil.wrapper
  File "h5py\h5f.pyx", line 106, in h5py.h5f.open
--------------------------------------------------------------------------------
Dioptas Version: 0.5.7
[image]

CPrescher, 2024-01-31:
Hi, very unclear from the message, but Dioptas seems to be not able to find the file information. Are you sure these hdf5 should be possible to load? Could you upload an example file. From where do you have the file?

rorajo195, 2024-01-31:
Hi Clemens. 

I am not very familiar with Dioptas so maybe it is just a beginner mistake.

I am uploading two files. Both from the same source. One is working, the other one Dioptas shows the error posted.
[examples.zip](https://github.com/Dioptas/Dioptas/files/14114601/examples.zip)

Thanks

rorajo195, 2024-02-02:
Problem solved. It seems that the files were corrupted. 

Thanks.

Jose Luis

closed 2024-02-02 by rorajo195

CPrescher, 2024-02-02:
hi,

thank you for informing us!