Dioptas/Dioptas #187: Created shortcut do not work on macos (at least)

t20100, 2024-02-12:
Running `dioptas makeshortcut` on macos creates a shortcut on the desktop that is not working.
This is because this call `/path/to/env/bin/python /path/to/env/bin/dioptas.py` but there is no `dioptas.py` available. I expect it to be the same at least on linux.

CPrescher, 2024-02-12:
The make shortcut option has been created by @newville. If I am not mistaken, he has also now created an extra library, to do this.

newville, 2024-02-14:
@CPrescher I'll investigate....

CPrescher, 2024-02-15:
Thank you. I guess the main issue is that "dioptas.py" in the main directory was renamed into "run.py" due to some problems I had with pyinstaller (if I remember correctly MacOSX cannot distinguish files from folder with the same name)