Dioptas/Dioptas #191: Windows intallation problem

shentasa, 2024-04-07:
Hello!
I'm trying to install Dioptas in Windows, I created a conda evironment with python 3.9 and I get this error executing this command "create_executable.bat" in Dioptas-0.5.9

  File "C:\Users\shent\Anaconda3\envs\dioptasenv\lib\site-packages\PyInstaller\building\build_main.py", line 1079, in build
    exec(code, spec_namespace)
  File "Dioptas.spec", line 57, in <module>
    fabio_hiddenimports = get_libs("fabio")
  File "Dioptas.spec", line 44, in get_libs
    __import__(module_name)
ModuleNotFoundError: No module named 'fabio'

Could you help me with it please?

Best,

Cynthia

shentasa, 2024-04-07:
I also tried to install via pip, and I get this error... after install visual studio build tools 2022....

      building 'CifFile.StarScan' extension
      error: Microsoft Visual C++ 14.0 or greater is required. Get it with "Microsoft C++ Build Tools": https://visualstudio.microsoft.com/visual-cpp-build-tools/
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for PyCifRW
Failed to build PyCifRW
ERROR: Could not build wheels for PyCifRW, which is required to install pyproject.toml-based projects

CPrescher, 2024-04-08:
Hi, 
I am puzzled why you run "create_executable". Please follow the guideline in the readme of the repository. Already created executables for windows can be downloaded from the releases, e.g. https://github.com/Dioptas/Dioptas/releases/tag/0.5.8

I hope this helps.

You can also do "pip install dioptas". But for this the Microsoft C++ Build Tools": https://visualstudio.microsoft.com/visual-cpp-build-tools/ are required

Best wishes,
Clemens

shentasa, 2024-04-08:
Dear Clemens,

Thank you for your reply. I installed the the program via "pip install
dioptas" in a python 3.9 environment, in my computer with Windows 11, 64
bits. The program could be opened this way, however, when I load the
calibration file ".poni", I get the following error:

<class 'ValueError'>:
invalid literal for int() with base 10: '2.1'
--------------------------------------------------------------------------------
  File
"C:\Users\Cyntia\xraylarch\envs\dioptasenv\lib\site-packages\dioptas\controller\CalibrationController.py",
line 600, in load_calibration
    self.model.calibration_model.load(filename)
  File
"C:\Users\Cyntia\xraylarch\envs\dioptasenv\lib\site-packages\dioptas\model\CalibrationModel.py",
line 553, in load
    self.pattern_geometry.load(filename)
  File
"C:\Users\Cyntia\xraylarch\envs\dioptasenv\lib\site-packages\pyFAI\geometry\core.py",
line 1341, in load
    poni = ponifile.PoniFile(data=filename)
  File
"C:\Users\Cyntia\xraylarch\envs\dioptasenv\lib\site-packages\pyFAI\io\ponifile.py",
line 62, in __init__
    self.read_from_file(data)
  File
"C:\Users\Cyntia\xraylarch\envs\dioptasenv\lib\site-packages\pyFAI\io\ponifile.py",
line 94, in read_from_file
    self.read_from_dict(data)
  File
"C:\Users\Cyntia\xraylarch\envs\dioptasenv\lib\site-packages\pyFAI\io\ponifile.py",
line 101, in read_from_dict
    version = int(config.get("poni_version", 1))
--------------------------------------------------------------------------------
Dioptas Version: 0.5.9

I tried the executable in Dioptas 0.5.8 through the link you sent me, in
this way I got the following problem :

--------------------------------------------------------------------------------
<class 'ValueError'>:
invalid literal for int() with base 10: '2.1'
--------------------------------------------------------------------------------
  File "dioptas\controller\CalibrationController.py", line 600, in
load_calibration
  File "dioptas\model\CalibrationModel.py", line 553, in load
  File "pyFAI\geometry\core.py", line 1341, in load
  File "pyFAI\io\ponifile.py", line 62, in __init__
  File "pyFAI\io\ponifile.py", line 94, in read_from_file
  File "pyFAI\io\ponifile.py", line 101, in read_from_dict
--------------------------------------------------------------------------------
Dioptas Version:
0.5.8--------------------------------------------------------------------------------

Can you help me with this?

Best,

Cynthia

jaimefdeoliveira, 2024-04-08:
I installed via pip and had the same error when opening or creating a configuration file.

2024-04-08, 13:02:36
--------------------------------------------------------------------------------
<class 'ValueError'>: 
invalid literal for int() with base 10: '2.1'
--------------------------------------------------------------------------------
  File "C:\Users\jaime\anaconda3\envs\didi1\lib\site-packages\dioptas\controller\CalibrationController.py", line 600, in load_calibration
    self.model.calibration_model.load(filename)
  File "C:\Users\jaime\anaconda3\envs\didi1\lib\site-packages\dioptas\model\CalibrationModel.py", line 553, in load
    self.pattern_geometry.load(filename)
  File "C:\Users\jaime\anaconda3\envs\didi1\lib\site-packages\pyFAI\geometry\core.py", line 1341, in load
    poni = ponifile.PoniFile(data=filename)
  File "C:\Users\jaime\anaconda3\envs\didi1\lib\site-packages\pyFAI\io\ponifile.py", line 62, in __init__
    self.read_from_file(data)
  File "C:\Users\jaime\anaconda3\envs\didi1\lib\site-packages\pyFAI\io\ponifile.py", line 94, in read_from_file
    self.read_from_dict(data)
  File "C:\Users\jaime\anaconda3\envs\didi1\lib\site-packages\pyFAI\io\ponifile.py", line 101, in read_from_dict
    version = int(config.get("poni_version", 1))
--------------------------------------------------------------------------------
Dioptas Version: 0.5.9

CPrescher, 2024-04-08:
From where did you get the poni file? Could you upload it here?

It might be that there is a version problem - possible solution is to create your own calibration within dioptas - process is described in the manual.

if this is not an option, you could just insert the parameters manually into the Dioptas GUI (just open the poni file with a text editor and you can see the parameters).

Best wishes,
Clemens

shentasa, 2024-04-09:
I got it from the Sirius synchrotron in Brazil. I have attached the
calibration file made in the lab with dioptas; they installed it in Linux.
It seems that here the program does not recognize the point as a decimal
separator; it changes the values to semicolons. Thank you again for your
help!

closed 2024-04-10 by CPrescher