Dioptas/Dioptas #195: fail to change color bar

sihengw, 2024-05-01:
once clicked the color bar to change the JCPDS, the error message prompts as follows:

Error information:
--------------------------------------------------------------------------------
2024-04-30, 17:13:57
--------------------------------------------------------------------------------
<class 'TypeError'>: 
arguments did not match any overloaded call:
  color(self, cg: QPalette.ColorGroup, cr: QPalette.ColorRole): argument 1 has unexpected type 'int'
  color(self, cr: QPalette.ColorRole): argument 1 has unexpected type 'int'
--------------------------------------------------------------------------------
  File "dioptas\controller\integration\OverlayController.py", line 214, in color_btn_clicked
--------------------------------------------------------------------------------
Dioptas Version: 0.5.7

CPrescher, 2024-05-01:
Hi,

0.5.7 is a relatively old version. Please update to a newer version: https://github.com/Dioptas/Dioptas/releases where this bug is fixed.

Best wishes,
Clemnes

closed 2024-05-01 by CPrescher

sihengw, 2024-05-01:
Thank you!!  It works now.