Dioptas/Dioptas #199: Merging with newer fabio capable to read multimodule Lambdas

lorcat, 2024-05-14:
Hi! 
_fabio_ has [recently](https://github.com/silx-kit/fabio/issues/565) incorporated a change enabling it reading multimodule detectors, e.g. Lambda's through [HDF5 VDS](https://docs.h5py.org/en/stable/vds.html).

Would it be possible to build the 0.6.0 with this addition?
Thank you!

CPrescher, 2024-05-14:
Hi,

As far as i can see, the changes have been merged into the repository, but there has been no new release of fabio since then yet. Can we wait for this, or do you need it timely? I can create a release for you for the beamtime, but would not feel good to have a general release without a new official fabio release.

Best wishes,
Clemens

lorcat, 2024-05-17:
Hi! I think that it makes sense to wait for the official fabio release.
I will try to make the local install to check operation first. Thanks!
Konstantin

CPrescher, 2025-06-10:
Is this now working correctly?

lorcat, 2025-06-10:
Hi Clemens!
Thanks for asking. Yes, bu may require a small adjustment - can we check if there is a summation of multiframe images if open with "add" checkbox? I think that our users would like to have it. Thanks!

CPrescher, 2025-06-10:
Hm, do you want to open multiple multiframe image and add them together, or just one file and it should add the individual frames?

lorcat, 2025-06-10:
> Hm, do you want to open multiple multiframe images and add them together,

Hi!
I suggest that it should be both when using "add" option:
1 when a single multiframe file is opened with add option - all frames
within this fill should be summed
2 when multiple multiframe files are selected - all frames should be summed
up

Both features are requested by dynamic compression users
The second feature is also requested by other beamlines which collect hd5
files with small beams - they need to sum up frames from a map for easier
data analysis.

Thank you!
best wishes,
          Konstantin