Dioptas/Dioptas #204: Failed import of OpenGL.platform.egl prevents dioptas to be started

SebRoLENS, 2024-07-17:
Dioptas cannot be started during Wayland session providing the following error:  

`Unable to import OpenGL.platform.egl.EGLPlatform: No module named 'OpenGL.platform.egl'
`
This happen using the executable for Linux (debian trixie). The python installation doesn't work as well.

SebRoLENS, 2024-07-17:
> Hi,
>
> could you try to install the following packages and then start it again?
>
> sudo apt-get install python3-opengl libgl1-mesa-dev libglib2.0-0
>
> Best wishes,
> Clemens

Hi,

thanks for the fast reply

all the packages are already installed and updated.

After trying again, I can confirm it works well on X11 but the issue
happens under wayland. I'll try to figure out updating this post if I
understand.

Thanks

Best
S. Romi

-- 

Dr. Sebastiano Romi

Postdoc

UniFi - Dipartimento di Chimica “Ugo Schiff”
Università degli Studi di Firenze
Via della Lastruccia 3, I-50019 Sesto Fiorentino, Firenze, Italy

LENS - European Laboratory for Non-Linear Spectroscopy
Structure, Dynamics, and Reactivity - High-Pressure Chemistry and Physics
Via Nello Carrara 1 - 50019 Sesto Fiorentino (Firenze)

📞 Tel. +39 055 457 2503 (Office - Room 75)
🔬 Tel. +39 055 457 2515 (Lab 30)
🔬 Tel. +39 055 457 2529 (Lab 29)

CPrescher, 2024-07-21:
Hi,

thank you. For me it works easily on latest linux mint and X11. I am unsure about what the problem is.

Best wishes,
Clemens

SebRoLENS, 2024-11-04:
Hello again. After updating Debian, the same issue I reported before is back. This time it happen with X11 session too. Below I copied what I can see from the terminal launching Dioptas. This happen with v6.0.0 and v6.0.1, as well as with some random older version I tried. Later I can try with a second Fedora machine.
 
`./Dioptas 
WARNING:root:[Errno 2] File o directory non esistente: '/home/seb/.local/bin/dioptas/_internal/qt_material/fonts/roboto'
Dioptas 0.6.0
qt.glx: qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QColorSpace(), profile  QSurfaceFormat::NoProfile)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QColorSpace(), profile  QSurfaceFormat::NoProfile)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QColorSpace(), profile  QSurfaceFormat::NoProfile)
Could not initialize GLX
Annullato (core dump creato)
` 

Best
S.Romi

closed 2026-03-04 by CPrescher