Dioptas/Dioptas #205: Dioptas cannot calibrate Ni data

worlddatong, 2024-07-18:
We found that it is not possible to calibrate Ni (Standard Reference Material) data using dioptas. Diopta will report an error of ''The calibrant does not contain enough d-spacings." Is there any way to solve this problem?

If necessary, I can provide Ni data and related parameters.

Thank you very much.
[image]

CPrescher, 2024-07-21:
Currently the calibration file for Ni is defined like this:

# Calibrant: Nickel (Ni)
# Face centered cubic cell a=3.5239 b=3.5239 c=3.5239 alpha=90.000 beta=90.000 gamma=90.000
# Ref: A. Taylor, J. Inst. Metals, 1950, 77, 585.
2.03450729 # (1, 1, 1) 8
1.76193500 # (2, 0, 0) 6
1.24587619 # (2, 2, 0) 12
1.06248678 # (3, 1, 1) 24
1.01725365 # (2, 2, 2) 8

Thus, it will only use the first rings (but it will work).
If you need more than 5 rings, you can add your own ".D" file into the dioptas/ressources/calibrants folder and restart Dioptas, the name can then be chosen in the calibrants list. The file is just expected to have a d spacing for each diffraction line on each line (everything after a # is considered a comment).

Best wishes,
Clemens

worlddatong, 2024-07-24:
Thank you very much! This really helps.

closed 2024-07-24 by worlddatong