Dioptas/Dioptas #206: Failed to open a tif file

andrefantoine, 2024-07-24:
Attempted to open a tif file and recieved this error message:

--------------------------------------------------------------------------------
2024-07-24, 11:55:18
--------------------------------------------------------------------------------
<class 'ValueError'>: 
2D array only are supported
--------------------------------------------------------------------------------
  File "dioptas\controller\CalibrationController.py", line 219, in load_img
  File "dioptas\model\ImgModel.py", line 151, in load
  File "dioptas\model\util\signal.py", line 64, in emit
  File "dioptas\model\util\signal.py", line 72, in _serve_listeners
  File "dioptas\model\util\signal.py", line 65, in emit
  File "dioptas\model\util\signal.py", line 77, in _serve_listeners
  File "dioptas\controller\CalibrationController.py", line 396, in plot_image
  File "dioptas\widgets\plot_widgets\ImgWidget.py", line 118, in plot_image
  File "dioptas\widgets\plot_widgets\ImgWidget.py", line 153, in auto_level
  File "dioptas\widgets\plot_widgets\utils.py", line 169, in get_range
  File "dioptas\widgets\plot_widgets\utils.py", line 122, in detector_gap_mask
--------------------------------------------------------------------------------
Dioptas Version: 0.6.1--------------------------------------------------------------------------------
2024-07-24, 11:58:14
--------------------------------------------------------------------------------
<class 'ValueError'>: 
unsupported image type
--------------------------------------------------------------------------------
  File "pyqtgraph\graphicsItems\ImageItem.py", line 866, in paint
  File "pyqtgraph\graphicsItems\ImageItem.py", line 583, in render
  File "pyqtgraph\graphicsItems\ImageItem.py", line 855, in _try_make_qimage
--------------------------------------------------------------------------------
Dioptas Version: 0.6.1

CPrescher, 2024-07-24:
Hi,
thank you for the report. Unfortunately, this error is not very conclusive. Are you sure the tif file is valid? Do you maybe use any file sharing and the file is corrupted? Is the Tif file maybe just 1 dimensional? Where did you get the file from?
Could you maybe upload the tif file for further inspection?

Best wishes,
Clemens

CPrescher, 2024-07-30:
Ok, thanks for letting me know. Do you want the tiff file to be 1 dimensional? I do not think Dioptas has a usecase for this kind of file:)

Best wishes,
Clemens