Dioptas/Dioptas #210: Versions 0.5.8. and 0.6.0 on Mac does not run

Bjoern20, 2025-02-11:
I use a Mac, and neither the Mac-available Dioptas version 0.5.8 nor 0.6.0 works. I only receive a simple error message from Finder stating that the program cannot be opened—without any detailed report. The process does not even reach the point of verifying the software or asking for the security prompts. If I change the permissions via the terminal using: `sudo chmod -R 755 Dioptas_0.6.0.app` the software is then checked, and I need to confirm the security prompts. However, after doing so, I receive a message stating that Dioptas quit unexpectedly, along with a crash report. See report attached: [report.txt](https://github.com/user-attachments/files/18748328/report.txt)

Version 0.5.7 works fine, but the issue where the „auto peak number keeps increasing despite the checkbox being unchecked“ is frustrating to work with, so I would like to update.

System Information:
- Device: MacBook Pro (2020)
- Operating System: macOS Sequoia 15.0
- Processor: 2.3 GHz Quad-Core Intel Core i7
- RAM: 16 GB
- Graphics Card: Intel Iris Plus Graphics (1536 MB)

A colleague with a MacBook Pro (2024) running macOS Sequoia 15.0 and an M3 Pro chip is experiencing the same issues.

Are there any ideas how to fix the issues? 

Best 
Björn

CPrescher, 2025-02-17:
Due to the ever changing policies of Mac OS X. I would recommend changing to a developer friendly operating system or run dioptas dioptas with a local python version. I am quite sure, this has worked for 0.6.0 on the latest macbook. (instructions are given in the main readme -either use pip or clone the repository and use poetry install)

CPrescher, 2025-06-10:
Does one of the newer versions under https://github.com/Dioptas/Dioptas/releases/tag/0.7.1 work?

CPrescher, 2025-06-12:
ok, I am happy to hear that.