Dioptas/Dioptas #213: an error message

springswim, 2025-03-05:
Please see the error message below and please help us on this issue. Thank you.

An unhandled exception occurred. Please report the bug under:

 	https://github.com/Dioptas/Dioptas/issues

or via email to:

	 <[clemens.prescher@gmail.com](mailto:clemens.prescher@gmail.com)>.

Please make sure to report the steps to reproduce the error. Otherwise it will be hard to fix it. 

A log has been written to "/home/xf28id1/dioptas_error.log".

Error information:

--------------------------------------------------------------------------------

2025-03-05, 16:52:34

--------------------------------------------------------------------------------

<class 'TypeError'>: 

PatternMatchingEventHandler.__init__() takes 1 positional argument but 2 were given

--------------------------------------------------------------------------------

  File "/home/xf28id1/conda_envs/dioptas/bin/dioptas", line 10, in <module>

    sys.exit(main())

             ^^^^^^

  File "/home/xf28id1/conda_envs/dioptas/lib/python3.11/site-packages/dioptas/__init__.py", line 63, in main

    controller = MainController()

                 ^^^^^^^^^^^^^^^^

  File "/home/xf28id1/conda_envs/dioptas/lib/python3.11/site-packages/dioptas/controller/MainController.py", line 65, in __init__

    self.model = DioptasModel()

                 ^^^^^^^^^^^^^^

  File "/home/xf28id1/conda_envs/dioptas/lib/python3.11/site-packages/dioptas/model/DioptasModel.py", line 56, in __init__

    self.configurations.append(Configuration())

                               ^^^^^^^^^^^^^^^

  File "/home/xf28id1/conda_envs/dioptas/lib/python3.11/site-packages/dioptas/model/Configuration.py", line 48, in __init__

    self.img_model = ImgModel()

                     ^^^^^^^^^^

  File "/home/xf28id1/conda_envs/dioptas/lib/python3.11/site-packages/dioptas/model/ImgModel.py", line 116, in __init__

    self._directory_watcher = NewFileInDirectoryWatcher(

                              ^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/home/xf28id1/conda_envs/dioptas/lib/python3.11/site-packages/dioptas/model/util/NewFileWatcher.py", line 68, in __init__

    self.event_handler = PatternMatchingEventHandler(self.patterns)

                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

--------------------------------------------------------------------------------

Dioptas Version: 0.6.1

In the message at terminal:

Qt: Session management error: None of the authentication protocols specified are supported
Dioptas 0.6.1

CPrescher, 2025-03-10:
hi,

thank you for sending the report. To test it, also need to have the following line be respected: "Please make sure to report the steps to reproduce the error." Could you please send the steps you do from starting dioptas to reproduce the error?

springswim, 2025-03-11:
At the beamline, we use Red Hat Enterprise Linux 8.10 (Oopta). The installation procedure is as follows:

1.	Open a terminal.
2. Create a Conda environment for Dioptas under /home/xf28id1/conda_envs/dioptas by typing: conda create -p /home/xf28id1/conda_envs/dioptas python”. We also tried with python=3.10 and python=3.13.
3. After creating a conda environment for dioptas, activate the Dioptas environment by typing: conda activate /home/xf28id1/conda_envs/dioptas.
4. Install Dioptas under /home/xf28id1/conda_envs/dioptas by typing: conda install -c conda-forge dioptas.

Please let me know if you have any questions. Thank you.

commit 56359985 (2025-03-21): fix: update to Watchdog >0.6.0 - change to new class initiator fixes #213
closed 2025-03-25 by CPrescher