Dioptas/Dioptas #214: Installing Dioptas: Encountered error while generating package metadata.

laakins, 2025-03-13:
I'm on Windows 11, Python 3.13.2, pip 25.0.1 and I'm very new to this environment.  I'm getting the following error when trying to install dioptas.  What am I doing wrong?

C:\Users\LAKINS>pip3 install dioptas
Collecting dioptas
  Using cached dioptas-0.5.8.tar.gz (1.1 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [49 lines of output]
      The "poetry.dev-dependencies" section is deprecated and will be removed in a future version. Use "poetry.group.dev.dependencies" instead.
      Traceback (most recent call last):
        File "C:\Users\LAKINS\AppData\Roaming\Python\Python313\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 389, in <module>
          main()
          ~~~~^^
        File "C:\Users\LAKINS\AppData\Roaming\Python\Python313\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 373, in main
          json_out["return_val"] = hook(**hook_input["kwargs"])
                                   ~~~~^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\LAKINS\AppData\Roaming\Python\Python313\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 175, in prepare_metadata_for_build_wheel
          return hook(metadata_directory, config_settings)
        File "C:\Users\LAKINS\AppData\Local\Temp\pip-build-env-lqpqc7bg\overlay\Lib\site-packages\poetry\core\masonry\api.py", line 42, in prepare_metadata_for_build_wheel
          poetry = Factory().create_poetry(Path().resolve(), with_groups=False)
        File "C:\Users\LAKINS\AppData\Local\Temp\pip-build-env-lqpqc7bg\overlay\Lib\site-packages\poetry_dynamic_versioning\patch.py", line 26, in alt_poetry_create
          name = _get_and_apply_version(
              pyproject_path=_get_pyproject_path_from_poetry(instance.pyproject),
          )
        File "C:\Users\LAKINS\AppData\Local\Temp\pip-build-env-lqpqc7bg\overlay\Lib\site-packages\poetry_dynamic_versioning\__init__.py", line 688, in _get_and_apply_version
          version, instance = _get_version(config, name)
                              ~~~~~~~~~~~~^^^^^^^^^^^^^^
        File "C:\Users\LAKINS\AppData\Local\Temp\pip-build-env-lqpqc7bg\overlay\Lib\site-packages\poetry_dynamic_versioning\__init__.py", line 502, in _get_version
          version = _get_version_from_dunamai(vcs, pattern, config)
        File "C:\Users\LAKINS\AppData\Local\Temp\pip-build-env-lqpqc7bg\overlay\Lib\site-packages\poetry_dynamic_versioning\__init__.py", line 463, in _get_version_from_dunamai
          return Version.from_vcs(
                 ~~~~~~~~~~~~~~~~^
              vcs=vcs,
              ^^^^^^^^
          ...<7 lines>...
              ignore_untracked=config["ignore-untracked"],
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
          )
          ^
        File "C:\Users\LAKINS\AppData\Local\Temp\pip-build-env-lqpqc7bg\overlay\Lib\site-packages\dunamai\__init__.py", line 2090, in from_vcs
          return cls._do_vcs_callback(
                 ~~~~~~~~~~~~~~~~~~~~^
              vcs,
              ^^^^
          ...<9 lines>...
              commit_length,
              ^^^^^^^^^^^^^^
          )
          ^
        File "C:\Users\LAKINS\AppData\Local\Temp\pip-build-env-lqpqc7bg\overlay\Lib\site-packages\dunamai\__init__.py", line 2145, in _do_vcs_callback
          return callback(**kwargs)
        File "C:\Users\LAKINS\AppData\Local\Temp\pip-build-env-lqpqc7bg\overlay\Lib\site-packages\dunamai\__init__.py", line 1104, in from_git
          _detect_vcs(vcs, path)
          ~~~~~~~~~~~^^^^^^^^^^^
        File "C:\Users\LAKINS\AppData\Local\Temp\pip-build-env-lqpqc7bg\overlay\Lib\site-packages\dunamai\__init__.py", line 371, in _detect_vcs
          raise RuntimeError("This does not appear to be a {} project".format(expected_vcs.value.title()))
      RuntimeError: This does not appear to be a Git project
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

VitanPy, 2025-03-17:
Hi, I'm trying to pip install Dioptas on my Mac and am getting the exact same issue... Hopefully this gets resolved soon :)

CPrescher, 2025-03-19:
To me the issue seems, that you are trying to install Dioptas from cache with version 5.8 - the current one is 6.2

Furthermore, do you need to use python 3.13? I cannot guarantee that this will work on Mac, but 3.12 should.

VitanPy, 2025-03-19:
Using the 'remove cache' results in this:
-s1928970@S44-4TQ7T60Y ~ % pip cache remove dioptas
WARNING: No matching packages for pattern "dioptas"
Files removed: 0 (0 bytes)

And attempting to download 0.6.1 does this:
-s1928970@S44-4TQ7T60Y ~ % pip install dioptas==0.6.1
Defaulting to user installation because normal site-packages is not writeable
ERROR: Could not find a version that satisfies the requirement dioptas==0.6.1 (from versions: 0.5.6, 0.5.7, 0.5.8)
ERROR: No matching distribution found for dioptas==0.6.1

If I then try to install 0.5.8:
-s1928970@S44-4TQ7T60Y ~ % pip install dioptas==0.5.8
Defaulting to user installation because normal site-packages is not writeable
Collecting dioptas==0.5.8
  Using cached dioptas-0.5.8.tar.gz (1.1 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [49 lines of output]
(and so on)

Vitan

laakins, 2025-03-19:
> To me the issue seems, that you are trying to install Dioptas from cache with version 0.5.8 - the current one is 0.6.1
> 
> The command to remove from the cache:
> 
> pip cache remove dioptas
> please make sure, that it is trying to install the latest version afterwards.
> 
> Furthermore, do you need to use python 3.13? I cannot guarantee that this will work on Mac, but 3.12 should.

> To me the issue seems, that you are trying to install Dioptas from cache with version 0.5.8 - the current one is 0.6.1
> 
> The command to remove from the cache:
> 
> pip cache remove dioptas
> please make sure, that it is trying to install the latest version afterwards.
> 
> Furthermore, do you need to use python 3.13? I cannot guarantee that this will work on Mac, but 3.12 should.

I'm running on Windows 11, not a Mac.  I'd like to be able to use python 3.13 because of vulnerabilities.  
I'm having similar problems as Vitan when removing the cache.

          Microsoft Windows [Version 10.0.22631.4890]
          (c) Microsoft Corporation. All rights reserved.
          
          C:\Users\LAKINS> pip cache remove dioptas
          WARNING: No matching packages for pattern "dioptas"
          Files removed: 0 (0 bytes)
          
          C:\Users\LAKINS>pip install dioptas==0.6.1
          ERROR: Could not find a version that satisfies the requirement dioptas==0.6.1 (from versions: 0.5.6, 0.5.7, 0.5.8)
          ERROR: No matching distribution found for dioptas==0.6.1
          
          C:\Users\LAKINS>

CPrescher, 2025-03-25:
If you're using Windows, you can simply download the pre-built executable as described in the README:
https://github.com/Dioptas/Dioptas/releases

Python 3.13 was released six months ago, but Python 3.12 will continue to be supported until 2028. There’s no need to worry about security vulnerabilities with Python 3.12 at this time. It's common for packages to take a while to support new Python versions, since many dependencies must also update to be compatible with the latest release.

I am currently working on Python 3.13 support, but I can't provide an estimated completion date yet.

For macOS users, please try using Python 3.12 for now.