Dioptas/Dioptas #216: Qt Platform Plugin

VitanPy, 2025-04-10:
Hi,

I'm on a Mac and am getting this issue when attempting to run Dioptas after installing it:

Last login: Thu Apr 10 15:59:49 on ttys000
(base) s1928970@S44-4TQ7T60Y ~ % dioptas
qt.qpa.plugin: Could not find the Qt platform plugin "cocoa" in ""
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

zsh: abort      dioptas
(base) s1928970@S44-4TQ7T60Y ~ % 

Reinstalling the application does not in fact fix the problem.

Vitan

CPrescher, 2025-04-10:
Hi,

please give some more information what you mean by install?

Are you using Anaconda or a normal python?
Do you use pip install or conda install?

Can you also try to use the Mac App under https://github.com/Dioptas/Dioptas/releases/tag/0.7.1

Best wishes,
Clemens

VitanPy, 2025-04-10:
I used 'pip install dioptas'. When I then try to call the application through the terminal, I get that error.

I just tried downloading the app from the link you provided above and after downloading the app just crashes and the error message says 'Dioptas 0.7.1 quit unexpectedly'. Reopening the app just recreates the error.

Vitan

CPrescher, 2025-04-10:
unfortunately, i do not have a mac to test things. 
You still did not tell me whether you are using normal python installation or anaconda, but from the terminal i suppose it is anaconda.
please try to use
conda install dioptas

the qt not found error is a common error, and might be specific to your python installation...
i suppose using pip in an anaconda environment is bound to not work.

VitanPy, 2025-04-10:
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               run_dioptas [11481]
Path:                  /private/var/folders/*/Dioptas_0.7.1.app/Contents/MacOS/run_dioptas
Identifier:            Dioptas_0.7.1
Version:               0.0.0 (???)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2025-04-10 19:17:27.2883 +0100
OS Version:            macOS 15.4 (24E248)
Report Version:        12
Anonymous UUID:        61C89B10-B918-F9B5-DEE1-A43D0EBB4190

Sleep/Wake UUID:       233D85CC-27F5-4E89-BEB1-AF2F83F96164

Time Awake Since Boot: 7300 seconds
Time Since Wake:       13 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000008
Exception Codes:       0x0000000000000001, 0x0000000000000008

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [11481]

VM Region Info: 0x8 is not in any region.  Bytes before following region: 4330078200
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      10217c000-102188000    [   48K] r-x/r-x SM=COW  /var/folders/*/Dioptas_0.7.1.app/Contents/MacOS/run_dioptas

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   CoreFoundation                	       0x1903c06cc __CFCheckCFInfoPACSignature + 4
1   CoreFoundation                	       0x190293548 CFBundleCopyBundleURL + 24
2   QtCore                        	       0x107e12aa8 QLibraryInfoPrivate::paths(QLibraryInfo::LibraryPath, QLibraryInfoPrivate::UsageMode) + 1960
3   QtCore                        	       0x107e12224 QLibraryInfoPrivate::path(QLibraryInfo::LibraryPath, QLibraryInfoPrivate::UsageMode) + 48
4   QtCore                        	       0x107e12d38 QLibraryInfoPrivate::paths(QLibraryInfo::LibraryPath, QLibraryInfoPrivate::UsageMode) + 2616
5   QtCore                        	       0x107e12224 QLibraryInfoPrivate::path(QLibraryInfo::LibraryPath, QLibraryInfoPrivate::UsageMode) + 48
6   QtCore                        	       0x107e54cb0 0x107df4000 + 396464
7   QtCore                        	       0x107e5411c 0x107df4000 + 393500
8   QtCore                        	       0x107e56798 0x107df4000 + 403352
9   QtCore                        	       0x107e531cc QLoggingCategory::QLoggingCategory(char const*, QtMsgType) + 52
10  QtCore.abi3.so                	       0x10761d604 warmUpLocationServices() + 300
11  QtCore.abi3.so                	       0x1076202a4 _GLOBAL__sub_I_qdarwinpermissionplugin_location.mm + 12
12  dyld                          	       0x18fe7ec18 invocation function for block in dyld4::Loader::findAndRunAllInitializers(dyld4::RuntimeState&) const + 444
13  dyld                          	       0x18febb0b4 invocation function for block in dyld3::MachOAnalyzer::forEachInitializer(Diagnostics&, dyld3::MachOAnalyzer::VMAddrConverter const&, void (unsigned int) block_pointer, void const*) const + 324
14  dyld                          	       0x18fedad70 invocation function for block in mach_o::Header::forEachSection(void (mach_o::Header::SectionInfo const&, bool&) block_pointer) const + 240
15  dyld                          	       0x18fed7b60 mach_o::Header::forEachLoadCommand(void (load_command const*, bool&) block_pointer) const + 208
16  dyld                          	       0x18fed92a0 mach_o::Header::forEachSection(void (mach_o::Header::SectionInfo const&, bool&) block_pointer) const + 124
17  dyld                          	       0x18febab84 dyld3::MachOAnalyzer::forEachInitializer(Diagnostics&, dyld3::MachOAnalyzer::VMAddrConverter const&, void (unsigned int) block_pointer, void const*) const + 516
18  dyld                          	       0x18fe7e9d0 dyld4::Loader::findAndRunAllInitializers(dyld4::RuntimeState&) const + 176
19  dyld                          	       0x18fe8635c dyld4::JustInTimeLoader::runInitializers(dyld4::RuntimeState&) const + 36
20  dyld                          	       0x18fe7f16c dyld4::Loader::runInitializersBottomUp(dyld4::RuntimeState&, dyld3::Array<dyld4::Loader const*>&, dyld3::Array<dyld4::Loader const*>&) const + 308
21  dyld                          	       0x18fe838dc dyld4::Loader::runInitializersBottomUpPlusUpwardLinks(dyld4::RuntimeState&) const::$_0::operator()() const + 180
22  dyld                          	       0x18fe7f478 dyld4::Loader::runInitializersBottomUpPlusUpwardLinks(dyld4::RuntimeState&) const + 700
23  dyld                          	       0x18fea2924 dyld4::APIs::dlopen_from(char const*, int, void*)::$_0::operator()() const + 1840
24  dyld                          	       0x18fe978c4 dyld4::APIs::dlopen_from(char const*, int, void*) + 1120
25  dyld                          	       0x18fe973b8 dyld4::APIs::dlopen(char const*, int) + 128
26  Python                        	       0x1043c2d00 _imp_create_dynamic + 888
27  Python                        	       0x104228020 cfunction_vectorcall_FASTCALL + 96
28  Python                        	       0x10434ee3c _PyEval_EvalFrameDefault + 59612
29  Python                        	       0x10433e058 PyEval_EvalCode + 132
[... 583 lines cut ...]
  "experiments" : [
    {
[... cut ...]
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4388), wl0: Feb 27 2025 18:17:26 version 23.40.26.0.41.51.177 FWID 01-36c62c6c
IO80211_driverkit-1475.34 "IO80211_driverkit-1475.34" Mar  9 2025 20:59:13
AirPort: 
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: USB31Bus
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.

VitanPy, 2025-04-10:
If I use conda install I get the following error:

Last login: Thu Apr 10 19:18:28 on ttys000
(base) s1928970@S44-4TQ7T60Y ~ % conda install dioptas
Channels:
 - conda-forge
 - defaults
Platform: osx-arm64
Collecting package metadata (repodata.json): done
Solving environment: failed

LibMambaUnsatisfiableError: Encountered problems while solving:
  - nothing provides pyshortcuts needed by dioptas-0.6.0-pyhd8ed1ab_0

Could not solve for environment specs
The following package could not be installed
└─ dioptas is not installable because it requires
   └─ pyshortcuts, which does not exist (perhaps a missing channel).

If I use pip install outside the conda environment I get the metadata error that I reported in the other GitHub issue post.

For some reason, using 'pip install dioptas' within the conda environment is the only way I could even install dioptas without issues. But then, when trying to open it, the above mentioned error occurs. I have no idea what's happening.

Vitan

CPrescher, 2025-04-11:
Hi, thank you for the detailed report:

Could you try, the solution under: https://stackoverflow.com/questions/76898551/qt-qpa-plugin-could-not-find-the-qt-platform-plugin-cocoa-in

for the qt platform error when using the pip install method

For the conda Installation - i am bit confused:

There is no reason it should install dioptas version 0.6.0 - the current version on conda forge is 0.7.1 - this i could also verify from 

```
conda install conda-forge:dioptas
```

under linux - it is a noarch package. Thus it should be available on any system. Also the pyshortcuts package is available on conda forge without issues. Are you using

CPrescher, 2025-04-11:
Can you please also try a new app file I generated, which should also set the qt plugin path correctly. You can download it from:
https://www.dropbox.com/scl/fi/a5g4pdw3a5c2i8pa5qbcr/DioptasMac_new.zip?rlkey=rye6afg3d8jhbq4tt600b17xy&st=o5mbjjn7&dl=0

Best wishes,
Clemens

VitanPy, 2025-04-11:
I used the provided link and the app still refuses to open. Here is the report:

I'll try to implement the conda and pip fixes you suggested also.

Vitan

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               run_dioptas [27985]
Path:                  /private/var/folders/*/Dioptas_0.7.1-post.3+29331703.app/Contents/MacOS/run_dioptas
Identifier:            Dioptas_0.7.1-post.3+29331703
Version:               0.0.0 (???)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2025-04-11 08:47:49.7528 +0100
OS Version:            macOS 15.4 (24E248)
Report Version:        12
Anonymous UUID:        61C89B10-B918-F9B5-DEE1-A43D0EBB4190

Sleep/Wake UUID:       AAD7B95C-BD4D-4FB5-998E-211CA8E8EB91

Time Awake Since Boot: 36000 seconds
Time Since Wake:       492 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000008
Exception Codes:       0x0000000000000001, 0x0000000000000008

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [27985]

VM Region Info: 0x8 is not in any region.  Bytes before following region: 4339744760
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      102ab4000-102ac0000    [   48K] r-x/r-x SM=COW  /var/folders/*/Dioptas_0.7.1-post.3+29331703.app/Contents/MacOS/run_dioptas

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   CoreFoundation                	       0x1903c06cc __CFCheckCFInfoPACSignature + 4
1   CoreFoundation                	       0x190293548 CFBundleCopyBundleURL + 24
2   QtCore                        	       0x106c02aa8 QLibraryInfoPrivate::paths(QLibraryInfo::LibraryPath, QLibraryInfoPrivate::UsageMode) + 1960
3   QtCore                        	       0x106c02224 QLibraryInfoPrivate::path(QLibraryInfo::LibraryPath, QLibraryInfoPrivate::UsageMode) + 48
4   QtCore                        	       0x106c02d38 QLibraryInfoPrivate::paths(QLibraryInfo::LibraryPath, QLibraryInfoPrivate::UsageMode) + 2616
5   QtCore                        	       0x106c02224 QLibraryInfoPrivate::path(QLibraryInfo::LibraryPath, QLibraryInfoPrivate::UsageMode) + 48
6   QtCore                        	       0x106c44cb0 0x106be4000 + 396464
7   QtCore                        	       0x106c4411c 0x106be4000 + 393500
8   QtCore                        	       0x106c46798 0x106be4000 + 403352
9   QtCore                        	       0x106c431cc QLoggingCategory::QLoggingCategory(char const*, QtMsgType) + 52
10  QtCore.abi3.so                	       0x10640d604 warmUpLocationServices() + 300
11  QtCore.abi3.so                	       0x1064102a4 _GLOBAL__sub_I_qdarwinpermissionplugin_location.mm + 12
12  dyld                          	       0x18fe7ec18 invocation function for block in dyld4::Loader::findAndRunAllInitializers(dyld4::RuntimeState&) const + 444
13  dyld                          	       0x18febb0b4 invocation function for block in dyld3::MachOAnalyzer::forEachInitializer(Diagnostics&, dyld3::MachOAnalyzer::VMAddrConverter const&, void (unsigned int) block_pointer, void const*) const + 324
14  dyld                          	       0x18fedad70 invocation function for block in mach_o::Header::forEachSection(void (mach_o::Header::SectionInfo const&, bool&) block_pointer) const + 240
15  dyld                          	       0x18fed7b60 mach_o::Header::forEachLoadCommand(void (load_command const*, bool&) block_pointer) const + 208
16  dyld                          	       0x18fed92a0 mach_o::Header::forEachSection(void (mach_o::Header::SectionInfo const&, bool&) block_pointer) const + 124
17  dyld                          	       0x18febab84 dyld3::MachOAnalyzer::forEachInitializer(Diagnostics&, dyld3::MachOAnalyzer::VMAddrConverter const&, void (unsigned int) block_pointer, void const*) const + 516
18  dyld                          	       0x18fe7e9d0 dyld4::Loader::findAndRunAllInitializers(dyld4::RuntimeState&) const + 176
19  dyld                          	       0x18fe8635c dyld4::JustInTimeLoader::runInitializers(dyld4::RuntimeState&) const + 36
20  dyld                          	       0x18fe7f16c dyld4::Loader::runInitializersBottomUp(dyld4::RuntimeState&, dyld3::Array<dyld4::Loader const*>&, dyld3::Array<dyld4::Loader const*>&) const + 308
21  dyld                          	       0x18fe838dc dyld4::Loader::runInitializersBottomUpPlusUpwardLinks(dyld4::RuntimeState&) const::$_0::operator()() const + 180
22  dyld                          	       0x18fe7f478 dyld4::Loader::runInitializersBottomUpPlusUpwardLinks(dyld4::RuntimeState&) const + 700
23  dyld                          	       0x18fea2924 dyld4::APIs::dlopen_from(char const*, int, void*)::$_0::operator()() const + 1840
[... 574 lines cut ...]
  "rollouts" : [
    {
      "rolloutId" : "645eb1d0417dab722a215927",
      "factorPackIds" : {

      },
      "deploymentId" : 240000005
    },
    {
      "rolloutId" : "661464ecda55e5192b100804",
      "factorPackIds" : {

      },
      "deploymentId" : 240000005
    }
  ],
  "experiments" : [
    {
      "treatmentId" : "ccf01f2d-382e-4ccc-a332-b3f688bc9bbf",
      "experimentId" : "67f46b1a73932228c9baf274",
      "deploymentId" : 400000001
    }
  ]
}
}

CPrescher, 2025-06-01:
I have uploaded a new executable app for arm64 architecture under: [](https://github.com/Dioptas/Dioptas/releases/tag/0.7.1)

Could you please try if this works for you?

CPrescher, 2025-06-01:
I also now tried to install dioptas on a fresh m1 Mac and it worked with two different ways:
- using the python from python.org
- installing newest miniconda from anaconda.org

and then just install it via: `pip install dioptas`
then running it with: `dioptas`

There is an issue with the conda-forge packages, they are currently only available on the old intel architecture and not for apple silicon arm64 architecture.

CPrescher, 2025-06-10:
any news on this?