Dioptas/Dioptas #217: Update Dioptas 0.7.1 not possible on my Mac

MH-marion, 2025-04-28:
Hi Clemens,

I could not install the last version of Dioptas on my Mac.
When using forge conda, I can only see the version 0.5.9. 
When downloading the version 0.7.1 and opening it I get the following message : "Impossible to open « Dioptas_0.7.1 » This is not taken care by your Mac." It doesn't show as well in the security options tab of Mac to "open the app anyway".
I am aware you don't have Mac to test it but is there any chance you know what to do?
If not, is it possible to update the forge condo version?

Many thanks for all your work!

Best

CPrescher, 2025-06-01:
Hi Marion,

which Mac are you using the new m (arm) architecture or old?
I just uploaded a new arm app file which might should work on the new architecture.

In any case the new operation system needs to follow these steps to overcome the "cannot be verified" warning:

1. After getting the cannot be verified warning:
  Click Cancel (not "Move to Bin").

2. Open System Settings > Privacy & Security.

3. Scroll down and you should see something like:

  "App was blocked from use because it is not from an identified developer."

4. Click Allow Anyway.

5. Try opening the app again  now macOS will say:

6. "Are you sure you want to open it?"

  Click Open  it will launch.

MH-marion, 2025-06-10:
Hi Clemens,
This is perfect. I updated your new version using Honda and it all worked perfectly!
I didn't have to give specific authorization. 
Many thanks! 

Marion