Dioptas/Dioptas #218: Issue with making cake from a valid image (Dioptas 0.7.1)

lorcat, 2025-06-06:
Hi!

Here is the log. DIOPTAS version 0.7.1 official Win build.
When switching from an 2D pattern view to a cake view within the integration widget, the following issue appeared:

2025-06-06, 14:00:07
--------------------------------------------------------------------------------
<class 'Exception'>: 
levels argument is required for float input types
--------------------------------------------------------------------------------
  File "pyqtgraph\graphicsItems\ImageItem.py", line 616, in paint
  File "pyqtgraph\graphicsItems\ImageItem.py", line 603, in render
  File "pyqtgraph\functions.py", line 1374, in makeARGB
--------------------------------------------------------------------------------
Dioptas Version: 0.7.1

The issue is annoying as it requires 

The file + calibration can be provided on a request.

CPrescher, 2025-06-10:
Hi Konstantin,

I remember this error, and thought I have fixed it. When I freshly download DIoptas 0.7.1 and run it on my own machine I also do not experience it. Does this happen consistently on your side? Could you provide your files so I can try to recreate it better?

Best wishes,
Clemens

lorcat, 2025-06-10:
Hi!
It is strange, but now I cannot reproduce it using the same images as
before and fresh dioptas start.
Maybe it was an accumulation of some errors. Sorry for bothering.
best wishes,
           Konstantin

CPrescher, 2025-06-10:
I could imagine that you where starting it from a python isntallation with an old pyqtgraph version - this would probably cause a similar error. I will close the issue for know. Feel free to open it in case it becomese reproducible again.

closed 2025-06-10 by CPrescher

lorcat, 2025-06-10:
Hi Clemens,
Well I was working with the 0.7.1 exe version downloaded from you at my
desktop, this I am pretty sure of.
Thanks!
best wishes,
            Konstantin